ROBOTIS-GIT/DynamixelSDK #458: Max number of dynamixels during sync write

ksmola, 2021-01-15:
C application on Embedded Linux, native RS485 port
I'm having trouble doing a sync write if there are more than 48 Dynamixels on the bus. (mix between PRO, PRO+ and Xseries) 
The drives respond OK to a sync read, but I can't command new goal positions with a sync write. Everything works fine when sending a sync write to <=48 drives. Anything above 48 fails. This is not related to their IDs, just the number of drives. 
Is this a known limitation and is there a workaround?

Thanks

ROBOTIS-Will, 2021-01-21:
Hi @ksmola 
I'm sorry about the delayed response.
I'll review this issue and get back to you as soon as possible.
Thank you.

ksmola, 2021-01-21:
daisy chained, about 2m, termination on both ends (120Ohm each), 2Mbit
Solved it by sending two sync writes back to back to keep the individual message length under 255.
Anything above 255 fails.
Thank you

ROBOTIS-Will, 2021-01-21:
The buffer in DYNAMIXEL is way larger than 255 and it shouldn't limit receiving the instruction.
Have you checked the TX data from the embedded board?
When I tested on my laptop with Wizard 2.0 using a modified c example `sync_read_write.c` from DYNAMIXEL SDK , I was able to see the correct packet transmission.
[image]

closed 2025-02-13 by yun-goon