ROBOTIS-GIT/DynamixelSDK #462: No module named 'dynamixel_sdk'

lara-z, 2021-01-25:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

**1. Which DYNAMIXEL SDK version do you use?**
  3.7.31

**2. Which programming language/tool do you use?**
  Python

**3. Which operating system do you use?**
  Windows 10

**3. Which USB serial converter do you use?**
  U2D2

**4. Which DYNAMIXEL do you use?**
   XL430-W250-T

**5. Have you searched the issue from the closed issue threads?**
  As best I can, yes

**6. Please describe the issue in detail**
This is my first time using dynamixel, and I cannot seem to get dynamixel sdk set up without producing a "No module named 'dynamixel_sdk'" error when running the protocol 2 test files.

I downloaded and installed R+ Task, Motion, and Manager 2.0. Since I want to control the motors with python, I downloaded dynamixel sdk from github and installed python (I was able to verify the python installation by running "python --version" in the command prompt). To install dynamixel sdk, I followed the following two robotis videos:

- https://www.youtube.com/watch?v=XMCPJFqoH10
- https://www.youtube.com/watch?v=pZmueNctY0s

When running setup.py according to one of the steps in the video, I seemed to have all the libraries already installed (though the guy in the video said I would see lots of libraries getting installed):
[image]

Despite this, I continued following the setup steps exactly and used the instructions in the second video to run the protocol 2.0 read_write.py test file. Upon running the file, I received the following error, which seems to say that I did not install dynamixel sdk:
[image]

I have repeated the processes in the videos multiple times with identical results. I also tried googling my issue and was unable to find a solution.

In terms of physical setup, I connected my motors (XL43-W250-T) to a U2D2 board and was able to see them via the R+ Wizard. So, I believe the physical setup is correct and should not have any impact on my problem.

**7. How can we reproduce the issue?**
By following the steps I described above.

ROBOTIS-David, 2021-01-26:
Hi, lara 

Seems that the library is not properly build from the source code. 

Please build the **setup.py** file by typing in the following command. 

```bash
python setup.py install
```

[image]

Make sure that the setup is properly progressed without no error messages in the resulting output. 

For more detailed setup, please go to the [DYNAMIXEL SDK - Python Windows Library Setup](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/python_windows/#python-windows) 

If you encounter any other issues, I will get back to you, 

Thank you,

lara-z, 2021-01-29:
Sorry, I forgot to reply when I read this, but it worked perfectly! Thank you so much!!

ROBOTIS-Will, 2021-02-01:
@lara-z 
Thank you for your feedback! Closing this thread.

closed 2021-02-01 by ROBOTIS-Will