ROBOTIS-GIT/DynamixelSDK #463: [XL430-W250-T][Reading issues][Incorrect status packet]

FrancoisCodeGen, 2021-02-01:
1. Which DYNAMIXEL SDK version do you use?
- last release
 
2. Which programming language/tool do you use?
  -  C
3. Which operating system do you use?
  - Ubuntu 64 bit

3. Which USB serial converter do you use?
  - U2D2 Power hub board and U2D2 USB/Series

4. Which DYNAMIXEL do you use?
  - five XL430-W250-T 

I am currently working on a joint robot with five XL430-W250-T. I'm driving my robot from my laptop thanks to an U2D2 USB/series. I use SDK sync_read_write function to write goal position and the motors respond perfectly. I also change PID values. 
I am working in 57600 bauds. 

Issues come when I am trying to sync_read present position from my motors. I wrote a program that reads the present position of the five servos 20 times per seconds and saves them in order to teach my robot a trajectory. 
It works for a few times but then suddenly the communication interrrupts and **TxRxResults** sends **Incorrect status packet**.
I can't figure out why... I tried to change the baud rates to go faster but without any success. 
Everything works perfectly except this problem of communication. Do you have any ideas or suggestions ?? 

thanks for your help !

ROBOTIS-Will, 2021-02-02:
Hi @FrancoisCodeGen 
I modified `sync_read_write.c` example as below and tried on two different settings.
- 5 x XL430-W250-T (ID: 1~5, Baudrate : 1Mbps, Return Delay Time : 250)
- 5 x XM430-W350-T (ID: 1~5, Baudrate : 1Mbps, Return Delay Time : 0)

My system environment is as below.
- Ubuntu 20.04 64bit with gcc 9.3.0
- DYNAMIXEL Starter Set (U2D2 + U2D2 PHB + 12V SMPS)
- DYNAMIXEL SDK lastest release
- /dev/ttyUSB0 port latency : 1

With this setup, I wasn't able to find any communication error.
Please try the code and let me know.

```c
#if defined(__linux__) || defined(__APPLE__)
#include <fcntl.h>
#include <termios.h>
#define STDIN_FILENO 0
#elif defined(_WIN32) || defined(_WIN64)
#include <conio.h>
#endif

#include <stdlib.h>
#include <stdio.h>
#include <time.h>
#include <errno.h>
#include "dynamixel_sdk.h"                                  // Uses Dynamixel SDK library

// Control table address
#define ADDR_PRO_TORQUE_ENABLE          64                 // Control table address is different in Dynamixel model
#define ADDR_PRO_GOAL_POSITION          116
#define ADDR_PRO_PRESENT_POSITION       132

// Data Byte Length
#define LEN_PRO_GOAL_POSITION           4
#define LEN_PRO_PRESENT_POSITION        4

// Protocol version
#define PROTOCOL_VERSION                2.0                 // See which protocol version is used in the Dynamixel

// Default setting
#define BAUDRATE                        1000000
#define DEVICENAME                      "/dev/ttyUSB0"      // Check which port is being used on your controller
                                                            // ex) Windows: "COM1"   Linux: "/dev/ttyUSB0" Mac: "/dev/tty.usbserial-*"

struct timespec tstart, tend;

int msleep(long msec)
{
    struct timespec ts;
    int res;

    if (msec < 0)
    {
        errno = EINVAL;
        return -1;
    }

    ts.tv_sec = msec / 1000;
    ts.tv_nsec = (msec % 1000) * 1000000;

    do {
        res = nanosleep(&ts, &ts);
    } while (res && errno == EINTR);

    return res;
}

int64_t timespecDiff(struct timespec *timeA_p, struct timespec *timeB_p)
{
  return ((timeA_p->tv_sec * 1000000000) + timeA_p->tv_nsec) -
           ((timeB_p->tv_sec * 1000000000) + timeB_p->tv_nsec);
}

int main(int argc, char **argv)
{
  int delay_value = 0;
  if (argc < 2) {
    printf("Usage : ./sync_read_write <delay_in_MS>\n");
    return 0;
  }

  delay_value = atoi(argv[1]);
  // Initialize PortHandler Structs
  // Set the port path
  // Get methods and members of PortHandlerLinux or PortHandlerWindows
  int port_num = portHandler(DEVICENAME);

  // Initialize PacketHandler Structs
  packetHandler();

  // Initialize Groupsyncread Structs for Present Position
  int groupread_num = groupSyncRead(port_num, PROTOCOL_VERSION, ADDR_PRO_PRESENT_POSITION, LEN_PRO_PRESENT_POSITION);

  int dxl_comm_result = COMM_TX_FAIL;               // Communication result
  uint8_t dxl_addparam_result = False;              // AddParam result
  uint8_t dxl_getdata_result = False;               // GetParam result

  uint8_t dxl_id[5] = {1, 2, 3, 4, 5};
  int32_t dxl_present_position[5] = {0};                           // Present position

  // Open port
  if (openPort(port_num))
  {
    printf("Succeeded to open the port!\n");
  }
  else
  {
    printf("Failed to open the port!\n");
    return 0;
  }

  // Set port baudrate
  if (setBaudRate(port_num, BAUDRATE))
  {
    printf("Succeeded to change the baudrate!\n");
  }
  else
  {
    printf("Failed to change the baudrate!\n");
    return 0;
  }

  // Add parameter storage for Dynamixel#1 present position value
  for(int i = 0; i < 5; ) {
    dxl_addparam_result = groupSyncReadAddParam(groupread_num, dxl_id[i]);
    if (dxl_addparam_result != True) {
      fprintf(stderr, "[ID:%03d] groupSyncRead addparam failed\n", dxl_id[i]);
      return 0;
    }
    else {
      printf("[ID:%03d] groupSyncRead addparam Succeeded\n", dxl_id[i]);
    }
    i++;
  }

  while (1)
  {
    clock_gettime(CLOCK_MONOTONIC, &tstart);
    // Syncread present position
    groupSyncReadTxRxPacket(groupread_num);
    if ((dxl_comm_result = getLastTxRxResult(port_num, PROTOCOL_VERSION)) != COMM_SUCCESS)
      printf("%s\n", getTxRxResult(PROTOCOL_VERSION, dxl_comm_result));

    // Check if groupsyncread data of Dynamixel#1 is available
    for (int j = 0; j < 5; ) {
      dxl_getdata_result = groupSyncReadIsAvailable(groupread_num, dxl_id[j], ADDR_PRO_PRESENT_POSITION, LEN_PRO_PRESENT_POSITION);
      
      if (dxl_getdata_result != True) {
        fprintf(stderr, "[ID:%03d] groupSyncRead getdata failed", dxl_id[j]);
        return 0;
      }
      else {
        dxl_present_position[j] = groupSyncReadGetData(groupread_num, dxl_id[j], ADDR_PRO_PRESENT_POSITION, LEN_PRO_PRESENT_POSITION);
        printf("[ID:%03d] :%03d  ", dxl_id[j], dxl_present_position[j]);
      }
      j++;
    }
    printf("\n");
    msleep(delay_value);
    clock_gettime(CLOCK_MONOTONIC, &tend);
    printf("[%ld ms] ", timespecDiff(&tend, &tstart)/1000000);
  }

  // Close port
  closePort(port_num);

  return 0;
}
```

FrancoisCodeGen, 2021-02-02:
Hi @ROBOTIS-Will 
Thanks for your answer. 

Unfortunately, I run your code and it's still not working... 

I set my port latency to 1 as you can see : 

[image]

I double cheked my baudrates and return delay time parameters. 
The weird thing is that some times one result shows up and i can read present positions of my servos as you can see : 

[image]

So obviously it's not a code issue or neither a parameter issue... 

One thing about my setup is that on my servo line, some of my wires are 35 cm wires (to reach joints of my robot). Do you think it could be an issue ? (Just to remember all my writting command programs are working fine) 

An other thing is that I am using a virtual linux machine wich runs on a windows10 laptop.  It might be also a part of the solution. 

Thanks again for your help.

ROBOTIS-Will, 2021-02-03:
@FrancoisCodeGen 
Hmm... Cables could be problematic, and replacing to a new one that came with DYNAMIXEL might help. 
At least we can cross out one of the most frequent cause of communication issues.
Since I haven't tested under the virtual machine, I cannot assure that the serial port of virtual machine works as that of Windows.

FrancoisCodeGen, 2021-02-03:
Hi again, 
 
I try today to run my code on a Raspberry pi and it works !!!
The issue was my virtual machine. 

Thanks for your help @ROBOTIS-Will .

ROBOTIS-Will, 2021-02-04:
Thanks for the updates!
Glad to hear that you found out where the issue came from.

yun-goon, 2025-02-13:
It looks like the issue has been resolved, so I will close it now.

closed 2025-02-13 by yun-goon