ROBOTIS-GIT/DynamixelSDK #464: ROS2 x Python

DLu, 2021-02-01:
1. Which DYNAMIXEL SDK version do you use?
3.7.31 (I believe)

2. Which programming language/tool do you use?
Python

3. Which operating system do you use?
Ubuntu 20.04

3. Which USB serial converter do you use?
unknown

4. Which DYNAMIXEL do you use?
XL430

5. Have you searched the issue from the closed issue threads?
Yes

6. Please describe the issue in detail
I would like to know if there are current plans to support the Python API in ROS2, particularly in Foxy/Galactic, or if there are particular reasons why they have not been supported yet. 

7. How can we reproduce the issue?
N/A

ROBOTIS-Will, 2021-02-22:
Hi @DLu 
Thank you for your inquiry.
The ROS2 examples will be updated within a couple weeks for C++ first, then followed by Python.

dschaele, 2023-07-20:
Hi @ROBOTIS-Will,
any news on Python support for ROS2? Just checked the humble and foxy releases and saw they are still featuring C++ only.

yun-goon, 2025-03-14:
Hello!

The ROS 2 Python functionality has been updated to the main branch! 🚀

Feel free to check it out, and let me know if you have any questions!

closed 2025-04-18 by yun-goon