ROBOTIS-GIT/DynamixelSDK #468: Dynamixel SDK - bulk read/write

ramonestana, 2021-02-10:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  - ex) `3.5.4`

2. Which programming language/tool do you use?
  - ex) `C#`,

3. Which operating system do you use?
  - ex) `Windows 10`,

3. Which USB serial converter do you use?
  - ex) `U2D2

4. Which DYNAMIXEL do you use?
  - ex) `XM540-W270-T.

5. Have you searched the issue from the closed issue threads?
    Yes
6. Please describe the issue in detail
I have tried to run the example BulkReadWrite with two XM540-W270-T, but nothing happend. In de terminal window, I get the following messages;

Succeeded to open the port!
Succeeded to change the baudrate!
Dynamixel1 has been successfully connected
Dynamixel2 has been successfully connected
Press any key to continue! (or press ESC to quit!)

But when I press a key, the messages at the screen are a bulk of equal messages like:
[ID: 1] Present Position : 415 [ID: 2] LED Value : 0
[ID: 1] Present Position : 415 [ID: 2] LED Value : 0
[ID: 1] Present Position : 415 [ID: 2] LED Value : 0
[ID: 1] Present Position : 415 [ID: 2] LED Value : 0
[ID: 1] Present Position : 415 [ID: 2] LED Value : 0

and so on. At the beginning, one of the servos is moving a little bit, but then nothing happens any more.

I should mention that the example with only one servo (example readwrite) is running perfectly (I changed the Control table values as you suggested in a previous thread).

Please help!

Thank you very much in advance!

Best,
Ramon

7. How can we reproduce the issue?

I dont know, sadly

ROBOTIS-David, 2021-02-16:
Hi, 

I am wondering if you could copy your code and paste it in order for me to look through. 

- Check if the Control Table address is properly set along with your DYNAMIXEL Model (XM430) in the class "BulkReadWrite"

ROBOTIS-Will, 2021-02-17:
@ramonestana 
Could you copy your `BulkReadWrite.cs` code here?
Without the source code, it is difficult for us to figure out what went wrong.
Thanks.

closed 2025-02-13 by yun-goon