ROBOTIS-GIT/DynamixelSDK #472: Dynamixel Simulink Library Error

wnghdwnsrl, 2021-03-10:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  - 3.7.4
  
2. Which programming language/tool do you use?
  - Simulink & Matlab

3. Which operating system do you use?
  - Windows 10

3. Which USB serial converter do you use?
  - U2D2

4. Which DYNAMIXEL do you use?
  - MX-64AT

5. Have you searched the issue from the closed issue threads?
   -Yes but there was no question or answer about Error in  Simulink. 

6. Please describe the issue in detail
   - I downloaded and used Dynamixel Simulink library , [https://kr.mathworks.com/matlabcentral/fileexchange/61944-dynamixel-simulink-library]
   - When I connected Mx-64 with mfile function myDxl = slDxl('COM7', 115200);
   It has no valid ID and num about MX-64. I just want to use it on Simulink. 
7. How can we reproduce the issue?
 Please, give me your comment.

ROBOTIS-Will, 2021-03-10:
Hi,
The DYNAMIXEL Simulink Library is a third party library that is not provided by ROBOTIS.
Please contact the developer for the issues related to the library.
Thank you.

johncgh22, 2021-03-13:
Hi @wnghdwnsrl 
I´m also trying to use DynamixelSDK with Simulink. So far, the one in MATLAB File Exchange doesn´t work for me. I get callib errors while using the myDxl function. I tried enabling **the headers and C functions** as mentioned here:

[https://la.mathworks.com/help/simulink/ug/integrate-ccode-ccaller.html](url)

In the mean time, I tried some methods to use the DynamixelSDK C library in Simulink. One of them is using **C Caller Blocks**. With this approach, I could get almost all the function needed to make a read_write example similar as the one available for LabVIEW, but i can´t get it to compile.

The second approach I used was with **shared libraries using MATLAB Function Block**, as metioned here:

[https://la.mathworks.com/help/simulink/ug/incorporate-c-code-using-a-matlab-function-block.html](url)
[https://la.mathworks.com/matlabcentral/fileexchange/44197-calling-shared-libraries-from-simulink](url)

But, I also having compile errors, and I basically stuck. If you wanna share how you configure the Library (the one made by the MATLAB team) and maybe get it to run, since i can´t get an answer from the original developer.

I'll be sharing my approaches if someone wants to get a look at them and try to get them work as a potential alternative.

Thanks in advance.

ROBOTIS-Shibata, 2021-03-14:
Hi,@johncgh22 @wnghdwnsrl 

Although it is not a material that can solve your issues directly, I will introduce Dynamixel related resources posted on MATLAB File Exchange.
They are wrapped as robot arm that using Dynamixel.

https://jp.mathworks.com/matlabcentral/fileexchange/65316-designing-robot-manipulator-algorithms
https://jp.mathworks.com/matlabcentral/fileexchange/67926-portrait-drawing-using-computer-vision-and-robot-manipulator

wnghdwnsrl, 2021-03-16:
Hi @johncgh22 
It was happened by small mistake.
I just changed  Devicename in "slDxl.m" file ( line 13)
`DeviceName = 'DeviceName';`
to
`DeviceName = 'myDxl';`

then it just worked.
I wish it can be helpful to you.

Hi @ROBOTIS-Shibata 
Thank you for your post.