ROBOTIS-GIT/DynamixelSDK #473: Dynamixel SDK examples : read_write.py // [TxRxResult] Incorrect status packet

judo0140, 2021-03-12:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  - ex) I don't know but , it's installed recently.

2. Which programming language/tool do you use?
  - ex) Python , ROS melodic

3. Which operating system do you use?
  - ex) `Ubuntu 18.04

3. Which USB serial converter do you use?
  - ex) OpenCR

4. Which DYNAMIXEL do you use?
  - ex) XM430-W210-T

5. Have you searched the issue from the closed issue threads?
 - Yes, According to the closed issue, I modified some parameters of read_write.py
6. Please describe the issue in detail
 - When I run read_write.py , an error message appears.

Succeeded to open the port
Succeeded to change the baudrate
[TxRxResult] Incorrect status packet!

The value of parameters is here

ADDR_PRO_TORQUE_ENABLE      = 64               
ADDR_PRO_GOAL_POSITION      = 116
ADDR_PRO_PRESENT_POSITION   = 132

# Protocol version
PROTOCOL_VERSION            = 2.0               

# Default setting
DXL_ID                      = 1                
BAUDRATE                    = 57600            
DEVICENAME                  = '/dev/ttyACM0'    

TORQUE_ENABLE               = 1                 
TORQUE_DISABLE              = 0               
DXL_MINIMUM_POSITION_VALUE  = 0        
DXL_MAXIMUM_POSITION_VALUE  = 4095            
DXL_MOVING_STATUS_THRESHOLD = 20  

7. How can we reproduce the issue?

ROBOTIS-Will, 2021-03-12:
Hi @judo0140 
Make sure that you are running a Protocol 2.0 read_write.py example.
In the latest example code, you only need to change the port from ttyUSB0 to ttyACM0
https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/python/tests/protocol2_0/read_write.py#L63

judo0140, 2021-03-12:
Thanks for your help.

I just found a problem on my OpenCR board (my board was reset.. so i had to reinstall usb_to_dxl)

So sorry...

closed 2021-03-12 by judo0140

ROBOTIS-Will, 2021-03-12:
@judo0140 
No worries at all and glad to hear that you found where to fix :)
Thanks for the updates!