ROBOTIS-GIT/DynamixelSDK #476: unable to make dynamixel SDK

bakwankawa, 2021-03-26:
help me fix this error:

mkdir -p .objects/
g++ -O2 -O3 -DLINUX -D_GNU_SOURCE -Wall -I../../../include/dynamixel_sdk -m64 -g  .objects/dxl_monitor.o -o dxl_monitor -ldxl_x64_cpp -lrt
/usr/bin/ld: cannot find -ldxl_x64_cpp
collect2: error: ld returned 1 exit status
Makefile:62: recipe for target 'dxl_monitor' failed
make: *** [dxl_monitor] Error 1

ROBOTIS-Will, 2021-03-29:
hi @bakwankawa 
In order to run the example, you need to build and install the c++ library first as described below.
https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/cpp_linux/#cpp-linux
Thanks!

bakwankawa, 2021-03-30:
> hi @bakwankawa
> In order to run the example, you need to build and install the c++ library first as described below.
> https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/cpp_linux/#cpp-linux
> Thanks!

Thanks for the solutions, work great!

closed 2022-02-09 by bakwankawa