ROBOTIS-GIT/DynamixelSDK #502: indirect_address example works with XM540 but not XL330?

ottensmeyer, 2021-05-28:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  - 3.7.31 (?) - was installed with OpenCR package for arduino around Jan, 2021

2. Which programming language/tool do you use?
  - Arduino IDE 1.8.13

3. Which operating system do you use?
  - Windows 10

3. Which USB serial converter do you use?
  - OpenCR

4. Which DYNAMIXEL do you use?
  - XM540-W150-T, XL330-M077-T

5. Have you searched the issue from the closed issue threads?
  - yes

6. Please describe the issue in detail
  - I am trying to use the indirect addressing example with the XM540 and XL330 servos (Protocol 2.0).  With the XM540, I change the address values in the #defines, change the position start/end locations to reasonable values (0, 1023), update the LED values (from 255 to 1), and I correctly read the position and moving state values
  - **When I do the same for the XL330, the position values are correctly read out, but the moving state never changes from 0.  I've replaced the moving state address (122) to other 1-byte addresses (e.g. temperature (146), LED state (65)), which also only read 0.**  
  - Writing using indirect address works fine - the goal_position and LED commands are accomplished successfully

  - Am I doing something wrong in using indirect addressing?  The example is for the Dynamixel PRO 54-200, and USB2DYNAMIXEL, and the example works correctly for the XM540 and OpenCR, but not the XL330
  - I've confirmed using the DYNAMIXEL Wizard 2.0 that the moving, temperature and other states are reading correctly from the XL330, which has address values for each of these the same as what's on the eManual (the Wizard also shows that the XL330 is not in need of an update)

  - Typical output using XM540 - PresPos is correctly reported, and IsMoving output shows the correct temperature (about 28) using the code below:

```
Start..
Succeeded to open the port!
Succeeded to change the baudrate!
Dynamixel has been successfully connected 
Press any key to continue! (or press q to quit!)
[ID:1] GoalPos:0  PresPos:371  IsMoving:28 
[ID:1] GoalPos:0  PresPos:364  IsMoving:28 
[ID:1] GoalPos:0  PresPos:346  IsMoving:28 
[ID:1] GoalPos:0  PresPos:324  IsMoving:28 
[ID:1] GoalPos:0  PresPos:302  IsMoving:28 
[ID:1] GoalPos:0  PresPos:280  IsMoving:28 
...
[ID:1] GoalPos:0  PresPos:34  IsMoving:28 
[ID:1] GoalPos:0  PresPos:28  IsMoving:28 
[ID:1] GoalPos:0  PresPos:23  IsMoving:28 
Press any key to continue! (or press q to quit!)

```

  - Typical output using XL330 - PresPos is correctly reported, but IsMoving output should be the temperature (about 28) using the code below, rather than 0:
```
Start..
Succeeded to open the port!
Succeeded to change the baudrate!
Dynamixel has been successfully connected 
Press any key to continue! (or press q to quit!)
[ID:2] GoalPos:0  PresPos:1016  IsMoving:0 
[ID:2] GoalPos:0  PresPos:1016  IsMoving:0 
[ID:2] GoalPos:0  PresPos:1013  IsMoving:0 
[ID:2] GoalPos:0  PresPos:995  IsMoving:0 
...
[ID:2] GoalPos:0  PresPos:41  IsMoving:0 
[ID:2] GoalPos:0  PresPos:31  IsMoving:0 
[ID:2] GoalPos:0  PresPos:26  IsMoving:0 
[ID:2] GoalPos:0  PresPos:23  IsMoving:0 
Press any key to continue! (or press q to quit!)
```
[... 377 lines cut ...]
  }

  // Disable Dynamixel Torque
  dxl_comm_result = packetHandler->write1ByteTxRx(portHandler, DXL_ID, ADDR_PRO_TORQUE_ENABLE, TORQUE_DISABLE, &dxl_error);
  if (dxl_comm_result != COMM_SUCCESS)
  {
    Serial.print(packetHandler->getTxRxResult(dxl_comm_result));
  }
  else if (dxl_error != 0)
  {
    Serial.print(packetHandler->getRxPacketError(dxl_error));
  }

  // Close port
  portHandler->closePort();

  return;
}

void loop()
{
  
}

```

ottensmeyer, 2021-05-30:
Update - tried with an XC430-W150-T - has the same indirect address/data addresses as the XM and works without problems.  No idea why the XL doesn't seem to work as expected.

ROBOTIS-Will, 2021-05-31:
Hi @ottensmeyer 
Thank you for your inquiry and sorry about your incovenience.
In the current XL330 firmware, there's a bug that some indirect address and data are not properly linked.
The next firmware update includes the fix for this bug.

ottensmeyer, 2021-05-31:
Thanks Will - glad it wasn't me doing something dumb. Do you have an ETA on the next update?

Cheers
Mark

ROBOTIS-Will, 2021-06-01:
Hi @ottensmeyer 
Thank you for your understanding.
Once the Wizard 2.0 software is updated, you'll get a notification.
The firmware update should be done manually as there are many customers who want to keep using a specific firmware version for their product.
Wizard 2.0 supports multiple module firmware update and you don't need to remove any connection from the system for a simple firmware update.
https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#firmware-update-all

The next update is scheduled in this month, but in case you urgently need to test the feature, please try the temporary firmware below.
Please note that this firmware fixes the indirect address & data bug and provided as is which does not guarantee its integrity.
https://drive.google.com/file/d/1a3gQKBToUtmg_l8UejzMILWscYeQwp7s/view?usp=sharing
Download will be available for the next 7 days.
Thank you!

ottensmeyer, 2021-06-01:
Hi Will - just installed the temporary fix - looks like it's working fine -
thanks very much for the early access - will keep my eyes out for the full
update!

best, Mark

yun-goon, 2025-02-04:
It looks like the issue has been resolved, so I will close it now.

closed 2025-02-04 by yun-goon