ROBOTIS-GIT/DynamixelSDK #503: make result not change

k-ish0212, 2021-05-31:
Hi

I'm new to Makefile and I tried to change PortHandlerLinux.cpp just add some printf()s, but the change was not reflected on the executable file. Otherwise some changes to example codes(like example/linux64/read_write/read_write.cpp) was reflected. Somebody knows Why the difference happen?

My Environment is Melodic and x64.

Thank you.

ROBOTIS-Will, 2021-06-01:
Hi @k-ish0212 
If you have modified the library, you should build and reinstall to make it effective.
Please refer to below eManual for building and reinstalling the library.
https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/cpp_linux/#build-the-library

k-ish0212, 2021-06-01:
thank you for your rapid reply, it worked!
I misunderstood it needs only $make clean.  I didn't $ sudo make reinstall

closed 2021-06-01 by k-ish0212