ROBOTIS-GIT/DynamixelSDK #504: No module named 'dynamixel_sdk'

Hello970, 2021-06-03:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  `3.7.31`

2. Which programming language/tool do you use?
  `Python`

3. Which operating system do you use?
  `Windows 10`

3. Which USB serial converter do you use?
  `U2D2`

4. Which DYNAMIXEL do you use?
  `MX-64AR(2.0)`

5. Have you searched the issue from the closed issue threads?
Yes, I checked some others met the same problem, but according to the answer given by ROBOTIS-David, the problem still cannot be solved (https://www.gitmemory.com/issue/ROBOTIS-GIT/DynamixelSDK/462/767229757).

6. Please describe the issue in detail
This is my first time using dynamixel with Python, and I cannot seem to get dynamixel sdk set up without producing a "No module named 'dynamixel_sdk'" error when running the protocol 2 test files and my own code. To install dynamixel sdk, I followed the following two robotis videos:

https://www.youtube.com/watch?v=XMCPJFqoH10
https://www.youtube.com/watch?v=pZmueNctY0s

I have repeated the processes in the videos multiple times with identical results. I also tried googling my issue and was unable to find a solution. As shown in attached figures 1 and 2, it seems the dynamixel_sdk has been installed, but when I try to test the code, it gives error " No module named 'dynamixel_sdk' ".
[image]
[image]
[image]

7. How can we reproduce the issue?
By following the steps I described above. Thanks for your attention.

ROBOTIS-Will, 2021-06-04:
Hi @Hello970 
Have you tried running the examples in the SDK?
Are you also getting the same error?
If the installed library cannot be found, make sure to add the library path to the path in windows environment.

Hello970, 2021-06-04:
Hi Sir, I have solved this problem by reinstalling Python. It might be caused by incorrect path. Thanks for your attention!

发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用

closed 2021-06-07 by ROBOTIS-Will