ROBOTIS-GIT/DynamixelSDK #505: fatal error: dynamixel_sdk/dynamixel_sdk.h: No such file or directory is:closed

abdul-mannan-khan, 2021-06-03:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  - ex) `3.4.1`

2. Which programming language/tool do you use?
  - `ROS`, `Arduino`

3. Which operating system do you use?
  - `Ubuntu 18.04`

3. Which USB serial converter do you use?
  - ex) `OpenCR`

4. Which DYNAMIXEL do you use?
  - `AX-12W'

5. Have you searched the issue from the closed issue threads? 
Yes.

6. Please describe the issue in detail
I am trying to install Dynamixel for linux. I am using OpenCR board. I am trying to follow [Setup DYNAMIXEL Workbench librar](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#opencr-and-opencm-tutorials) as described. However, when I run 

```
$ cd ~/dynamixel-workbench/dynamixel_workbench_toolbox/examples
$ mkdir -p build && cd build
$ cmake ..
$ make
```
I get the following error. 
`dynamixel_driver.h:29:12: fatal error: dynamixel_sdk/dynamixel_sdk.h: No such file or directory`

I don't know how to resolve this. I am simply following the steps given [here](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#opencr-and-opencm-tutorials). Any help will be appreciated. Thank you. 

7. How can we reproduce the issue?

ROBOTIS-Will, 2021-06-04:
The OpenCR board manager contains DYNAMIXEL Workbench library for OpenCR so you don't need to install a separate DYNAMIXEL Workbench on your PC.
Please follow the instructions below to install the OpenCR board manager for Arduino IDE.
https://emanual.robotis.com/docs/en/parts/controller/opencr10/#porting-to-arduino-idelinux

In case of running OpenCR as a communication interface between DYNAMIXEL and PC, it is recommended to use U2D2 in the DYNAMIXEL Starter Set instead.

vondar, 2021-07-16:
I have the same issue but, I'm using USB2DYNAMIXEL and ROS, my motors are AX-12A

ROBOTIS-Will, 2021-07-19:
@vondar 
You need to setup the DYNAMIXEL SDK library first as written in the eManual below.
https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#linux-and-macos

closed 2021-07-19 by abdul-mannan-khan