ROBOTIS-GIT/DynamixelSDK #506: Reading Load and Velocity from Dynamixel Servo

jeremyrhyde, 2021-06-09:
ISSUE TEMPLATE ver. 1.2.0[image]
Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  - 3.7.51

2. Which programming language/tool do you use?
  - Python

3. Which operating system do you use?
  - MacOS Catalina

3. Which USB serial converter do you use?
  - USB2Dynamixel

4. Which DYNAMIXEL do you use?
  - XL430-W250-T

5. Have you searched the issue from the closed issue threads?
- Yes

6. Please describe the issue in detail
- When attempting to read the velocity and load levels of my XL430-W250-T servo I get values that do not correlate to either and are often just nonsense. I'm trying to include an additional feedback mechanism by sensing these values, however, when I perform the readBytesTxRx (adjusting for num of bytes) on these I get numbers that don't correlate to the load or velocity present in the system. Both the position and temperature are working fine using the same method.

For load, I am using the 126 register and 2 bytes as prescribed by the https://emanual.robotis.com/docs/en/dxl/x/xl430-w250/ 
For velocity, I am using the 128 register 4 bytes from the same source

[image]

[image]

[image]

7. How can we reproduce the issue?
- attempt to read the values using the same registers and byte values as I do above. I can also share my .py files if desired (they were unsupported when I tried initially to upload)

ROBOTIS-Will, 2021-06-10:
@jeremyrhyde 
The Present Load(126) of XL430-W250 is a 2byte data so you should read it with read2ByteTxRx().
You might also need to properly print the unsigned value to signed value as Present Load could be a negative value.
Thank you.

jeremyrhyde, 2021-06-10:
My apologies, I realized my screenshot above had the read1ByteTxRx, I originally had it as 2Bytes but was temporarily experimenting with with 1Byte. The signed value did the trick though, thank you for the help!

closed 2021-06-10 by jeremyrhyde