ROBOTIS-GIT/DynamixelSDK #507: Writing registers works but reading does not

WormEater, 2021-06-09:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  - `3.4.1`

2. Which programming language/tool do you use?
  - `C`

3. Which operating system do you use?
  - Debian GNU/Linux 9

3. Which USB serial converter do you use?
  -  `U2D2`

4. Which DYNAMIXEL do you use?
  - `MX-12w`

5. Have you searched the issue from the closed issue threads?
  - `yes`
  
6. Please describe the issue in detail
  I am using a BeagleBoneBlue board to control the MX-12W motor. I first ran the sample code 'read_write' under linux_sbc folder, the motor operated as expected in multi-turn mode, which means the writing and reading functions worked properly. Then I implemented the functions to my own code. I was able to control the motor position which means the write functions works. However, I cannot get the position feedback from the motor.  The returned number was 0 and the returned message was '[TxRxResult] There is no status packet!'    

I connect beaglebone to U2D2 with the USB port. There are some other UART readings in parallel. I thought there might be some conflicts so I disabled the other threads and tested again. The result remained same: can write but cannot read.  

7. How can we reproduce the issue?
I'd attached the code below, but it is literally copying and pasting the sample code to my own code. the class name is xm430 but I changed necessary register addresses for mx-12w motor.
[image]

ROBOTIS-Will, 2021-06-10:
@WormEater 
I don't see any distinct error in your code except passing the `int pos` to `write2ByteTxRx` which should be automatically converted into uint16_t.
Could you copy the getLastTxRxResult() block and paste it after the write2ByteTxRx and see if writing data to the Goal Position works well?
Does your code work without xm430_read_position()?
I'd recommend try reading the Present Position after confirming wrte2ByteTxRx work.

WormEater, 2021-06-10:
@ROBOTIS-Will 
Thank you for your reply. 
- The writing data to Goal Position function operated properly. But the getLastTxRxResult() block returns '[TxRxResult] There is no status packet!' as well. This is interesting. 
- My code worked both with and without xm430_read_position(). When the xm430_read_position() was included, the program went through  the function and return 0.

WormEater, 2021-06-10:
@ROBOTIS-Will 

Additional information:
- I got the reading function working by directly call the function in main(). Before I placed the  function in a pthread and it never worked. 
- The writing function worked but return '[TxRxResult] There is no status packet!' probably due to the same reason. 
- I didn't check the lower level codes, there might be some conflicts between the pthread library and Dynamixel SDK.  
- In general I've solved my problem. Hopefully my information would be helpful if you'd like to dig more into this problem.

ROBOTIS-Will, 2021-06-10:
@WormEater 
If you have are running the reading thread separately, please check whether the read_position function instruction collides with the write instruction.
Since DYNAMIXEL will try to respond and return the status packet to almost all instructions, reading from DYNAMIXEL when writing to DYNAMIXEL won't work.

If you have an additional U2D2, you could use it as a "packet sniffer" via [Packet feature of DYNAMIXEL Wizard 2.0](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#packet).
This could help you to identify which instruction is / isn't working.

yun-goon, 2025-02-13:
Closing this thread as there isn't recent activity. Please feel free to reopen when necessary.

closed 2025-02-13 by yun-goon