ROBOTIS-GIT/DynamixelSDK #509: [3.7.31][U2D2][XH540-V270] goal current not reached

jstockermann, 2021-06-21:
1. Which DYNAMIXEL SDK version do you use?
  - 3.7.31

2. Which programming language/tool do you use?
  - MATLAB

3. Which operating system do you use?
  - Windows 10

3. Which USB serial converter do you use?
  - U2D2

4. Which DYNAMIXEL do you use?
  - XH540-V270

5. Have you searched the issue from the closed issue threads?
  - yes

6. Please describe the issue in detail
I use the XH540-V270 as a driver for an elastic and very lightweight arm. I try to control the motor position by commanding a current (torque control mode). In a loop, the desired current is calculated and sent via chinging the goal current of the robot. After that, I read the real current from the motor.
Unfortunately, the motor reaches the commanded current only when stalled, not in free motion. Is this some kind of safety feature, due to the low weight/resistance of the arm? (Changing the maximum velocity didn't seem to have any effect)

when stalled:
[image]

free motion:
[image]
(not in all experiments the real current as constant as in this image, if helpful I can upload another example)

7. How can we reproduce the issue?

ROBOTIS-Will, 2021-06-23:
Hi @jstockermann 
To give you the answer first, the current mode only uses a necessary current needed.
For example, the required force to lift a 100g object is smaller than that of a 1,000g object.
And you cannot use 1,000g object lifting force to lift a 100g object, but end up using only 100g lifting force.
If there's no load at the horn, there's no way to draw more current(force) from the motor.
Hope this explanation helps to understand the Current Mode of DYNAMIXEL.
Thanks!

jstockermann, 2021-06-23:
first: Thanks for your reply!
regarding your example: I may misunderstand something, but I would say that I can apply the same force to an object with 100 or 1000g, resulting in different accelerations. Likewise, I would assume that in my case the same current/torque, applied on an arm with a lower moment of inertia, results in a bigger rotational acceleration.
On the other hand, I get your point that with no load no current can build up in the motor (assuming that the motor itself has a moment of inertia of 0).

> the current mode only uses a necessary current needed.
What exactly defines what current is needed? 

Best regards, Jakob

ROBOTIS-Will, 2021-06-24:
@jstockermann 

In case you want to control the acceleration and velocity of the arm, you should use the Current-based Position Control mode.
https://emanual.robotis.com/docs/en/dxl/x/xh540-v270/#profile-acceleration108
https://emanual.robotis.com/docs/en/dxl/x/xh540-v270/#profile-velocity112

As described in the eManual, the Current Control mode is "_ideal for a system that only uses current control or a system that has additional velocity/position controllers_".

The required current is decided by the current flowing in the core motor.
As the core motor draws current to rotate the load, the current controller and limiter will provide or limit the current flowing into the core motor.

Thank you.

jstockermann, 2021-06-25:
thanks for your answers, I'll use another operating mode
Jakob

closed 2021-06-25 by jstockermann