ROBOTIS-GIT/DynamixelSDK #510: [Python 2.7][XL320][OpenCM9.04][Raspberry Pi Zero] Problems using OpenCM9.04 as Serial passthrough

Benoit-LdL, 2021-07-01:
1. Which DYNAMIXEL SDK version do you use?
  - 3.7.51

2. Which programming language/tool do you use?
  - Python 2.7

3. Which operating system do you use?
  - Raspbian 10 Buster

3. Which USB serial converter do you use?
  - OpenCM9.04

4. Which DYNAMIXEL do you use?
  - XL320

5. Have you searched the issue from the closed issue threads?
yes, I haven't been able to find anyone with the same idea as me.

6. Please describe the issue in detail
Hi, I have a Robotis Mini (XL320 servos) robot running an OpenCM9.04 board. I saw in the OpenCM9.04 documentation that the serial pins for the communication with the Dynamixel servos can be accessed using pin 11 and 12 on the board. I am running ROS Melodic on a Raspberry Pi Zero W running Raspbian 10 (Buster) and I thought it would be nice to control the servos using the raspberry Pi Zero W instead of the OpenCM9.04 board (using python = no compiling = quicker development). I have connected the Serial pins from the Pi to the Dynamixel serial pins 11 and 12 on the OpenCM9.04 (I do use a level shifter to step up/down the serial voltage). I have been able to send/receive normal data to/from both controllers. When I try the Dynamixel SDK read_write.py  script (or any other script from the SDK python protocol 1.0 directory) it tells me it successfully opened the port and changed the baudrate but no status packets are received...
 
```
pi@raspberrypi:~/DynamixelSDK/python/tests/protocol1_0 $ python read_write.py 
Succeeded to open the port
Succeeded to change the baudrate
[TxRxResult] There is no status packet!
Press any key to continue! (or press ESC to quit!)
```

7. How can we reproduce the issue?
### Hardware:
  - Raspberry Pi Zero W
  - OpenCM9.04
  - XL320 servos

#### Connections
| Pi Zero          | 4 channel bi-directional level shifter | OpenCM9.04  |
|------------------|----------------------------------------|-------------|
| UART TX (GPIO14) |-> | RXD_1 (D12) |
| UART RX (GPIO15) | <- | TXD_1 (D11) |

### Software:
 - Raspberry Pi Zero W: Raspbian Buster (10) running ROS Melodic
    - I use /dev/serial0  
 - OpenCM9.04 on which an empty Arduino sketch has been uploaded
    - If I'm not mistaken serial1 is connected to the servos and can be accessed using pins D11 and D12 on the OpenCM9.04 board.

8. What is my actual question?

Is this idea even possible, if not, could you explain to me why and if it is possible, how can I resolve my issue? 
A solution could be to buy a Raspberry Pi Zero hat specifically designed to communicate with Dynamixel servos or buy one of the many USB serial converter Robotis makes, but I would like to try this with the hardware I already own.

I hope my explanation is clear and I hope someone can help me with this ;)

Kind regards,

Benoît

ROBOTIS-Will, 2021-07-02:
Hi, 

Those two pins (D11, D12) are tied to the [JTAG/SWD port](https://emanual.robotis.com/docs/en/parts/controller/opencm904/#jtagswd-4-pin) and has not connected to any DYNAMIXEL ports.
If you wish to access to XL320 on OpenCM9.04, you should use [DYNAMIXEL 3Pin(TTL) port](https://emanual.robotis.com/docs/en/parts/controller/opencm904/#dynamixel-ttl-3-pin)

Benoit-LdL, 2021-07-03:
Hi, thanks for the quick reply!
However I'm not sure I understand  your response...

When looking at the OpenCM9.04 documentation on the emanual.robotis.com website, we can see this:
[image]
Name | Arduino Pin |  
-- | -- | --
Serial1 | D11, D12 | DXL Port

As far as I can see, the DXL port is connected to the Serial1 which can be accessed from pin D11 and D12.

While the JTAG/SWD port you're mentioning (4.3.10 in the documentation) are these pins:
[image]

While I do understand the logic of using the TTL ports to communicate with the XL320 servos, Both the Dynamixel SDK (python and arduino) and the Dynamixel workbench library for arduino use a serial connection to controll the servos

Dynamixel SDK Python
```python
# Default setting
DXL_ID                      = 1                             # Dynamixel ID: 1
BAUDRATE                    = 1000000
DEVICENAME                  = "/dev/serial1".encode('utf-8')
```

Dynamixel SDK Arduino
```c++
// Default setting
#define DXL_ID                          1                   // Dynamixel ID: 1
#define BAUDRATE                        1000000
#define DEVICENAME                      "1"                 // Check which port is being used on your controller
                                                                              // ex) Windows: "COM1"   Linux: "/dev/ttyUSB0"
```

Dynamixel workbench Arduino
```c++
#if defined(__OPENCM904__)
  #define DEVICE_NAME "3" //Dynamixel on Serial3(USART3)  <-OpenCM 485EXP
```

Maybe I did not understand your answer correclty or maybe my explanation was not clear enough, sorry for that!

I hope you can help me with this.
Kind regards,

Benoît

ROBOTIS-Will, 2021-07-05:
Hi @Benoit-LdL 

I've reviewed the connection with the PCB Schematics available from the [Download](https://emanual.robotis.com/docs/en/parts/controller/opencm904/#downloads) section of OpenCM9.04 and confirmed that the D11/12 are not connected to DXL Port.
[image]

It looks like the eManual is incorrectly describes the Serial1, which isn't assigned for any Arduino GPIO Pin.

The correct description will be :
|Name| Arduinio Pin| Description|
|:---:|:---:|:---:|
|Serial1 | N/A | DXL Port |
|Serial2 | A4, A5 | Communication Port(4Pin UART) |
|Serial3 | D24, D25 | OpenCM485 EXP DXL Port |

You may access to DYNAMIXEL modules that are connected to OpenCM9.04 via `DXL` GPIO port (tied to Serial1) as below.
[image]

In regards with the JTAG / SWD, I'm sorry about the wrong information. 
D11/D12 are tied to the IAP port on the other side of the JTAG / SWD for bootloader programming.

Thank you for pointing out the issue and we'll revise the information.
Have a great day!

Benoit-LdL, 2021-07-05:
Hi @ROBOTIS-Will ,

Thanks for the detailed explanation!
I will try to use the DXL GPIO pins as you described and keep you updated on my results.

Kind regards,
Benoît Lagasse

Benoit-LdL, 2021-07-05:
Hi @ROBOTIS-Will,
I just tried connecting DXL TX to Pi RX and DXL RX to Pi TX with no success. I'm still getting
```python
[TxRxResult] There is no status packet!
```
I did see in the Dynamixel Shield documentation that the D-pin of the DXL port is called DXL_DIR(DXL_TX_EN).
Pin No. | Pin Name | Description
-- | -- | --
0 | HW UART RX | DXL_RX
1 | HW UART TX | DXL_TX
2 | HW UART DIR | **DXL_DIR(DXL_TX_EN)**

I guess that this pin has to be put HIGH when transmitting data but I'm not sure...

Is there any documentation on using the DXL port directly to communicate and control servos? If so, where can I find this?
And can I use the Dynamixel SDK and other libraries to communicate directly with the DXL port?

Kind regards,

Benoît Lagasse

ROBOTIS-Will, 2021-07-06:
@Benoit-LdL 
Since you are controlling DYNAMIXEL only through RPi, OpenCM9.04 will only be considered as a power supply so no sketch would be needed.

I noticed that you are running the Protocol 1.0 example for XL-320, but XL-320 works with Protocol 2.0.
There are some minor modification needed in terms of size of data.
For example, usual Protocol 2.0 product like XL/XM/XH series have different control table structure than XL-320 and you should apply these differences in the example code.
Luckily, we've prepared a read_write.py example with an option to select DYNAMIXEL.
Simply commenting out the unused model, and activate the `MY_DXL = 'XL320'` definition.
https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/python/tests/protocol2_0/read_write.py#L55

Unfortunately, our examples are written based on the official products that we provide, and custom GPIO connections are hard to get supported.
If possible, could you give me a bit more details of your current project?

Benoit-LdL, 2021-07-06:
Hi @ROBOTIS-Will ,

I did not realise the xl320's work with protocol 2.0, sorry about that 😅

I will tinker a bit more on it now that you gave me some additional information!

If I still struggle to make it work, I will give a detailed explanation of what I'm trying to achieve in this project (I am going on vacation the following weeks so I may not answer for a while). 

Thanks for all the help until now! 

Kind regards,
Benoît Lagasse

ROBOTIS-Will, 2021-07-07:
@Benoit-LdL 
Sounds good.
Have a wonderful vacation and let me know if you have difficulty in running it.
Thanks!

Benoit-LdL, 2021-08-15:
Hi @ROBOTIS-Will,
I've tried to use the protocol2.0 read_write example (changed values to work with XL320 and use serial0 as port) but as you mentioned it was expected not to work when using custom GPIO's.

My goal is to make the Robotis Mini controllable through ROS. I've already tried using the Arduino library "rosserial" and I was able to control it via the usb interface of the OpenCM9.04. However, I found this solution not ideal because you need to re-compile the arduino code each time you change something. This is why I am looking at a more beginner-friendly way to program and control the robot using ROS (via python for example).

Because the Mini has a really low payload, I want to use a Pi Zero for this project. I have been able to get ROS running on it so the only thing left is to bridge the communication between the Pi and the servos.

I've seen some options online to make it work, such as [this Pi Zero Hat](https://www.tindie.com/products/savageelectronics/raspberry-pi-zero-dynamixel-hat/) or [this small interface board](https://www.tindie.com/products/savageelectronics/dynamixel-interface-board-v2/).

But before investing in such a board I wanted to see if there was not a way to use the hardware I've got to make it work. The OpenCm9.04 came with the robot and most pins (especially the DXL port) are accessible via the physical pinouts.

Is there a way to make this work without to much hassle, or should I just buy a external board?

Kind regards,

Benoît

ROBOTIS-Will, 2021-08-17:
Hi @Benoit-LdL 

It seems like using OpenCM9.04 as an interface board between the RPi and DYNAMIXEL will require some level of Arduino programming as OpenCM9.04 pins are controlled by the MCU.
I thought using communication ICs as passive devices would work, but wasn't successful in my case with an empty sketch.

Another possible way to do this is connecting RPi and OpenCM9.04 via `Serial3` while XL-320 is connected to the DXL  ports.
This still requires data transfer between `Serial3` and `Serial1` in Arduino code level, but could be a feasible solution.

Since this doesn't seem related to the issue of DYNAMIXEL SDK, it would be great if you can keep updating the status on ROBOTIS Community below so that other developers who are interested in this can participate.
https://community.robotis.us/

You might be also interested in `DYNAMIXEL Hat for Raspberry Pi`. 
Please contact at contactus2@robotis.com and check if there's any available beta test unit you can try.

Thank you.

Benoit-LdL, 2021-08-17:
Hi @ROBOTIS-Will,
Thanks for the intensive research you have done! I will indeed try out the Serial1-Serial3 option and if I have any success using that method I will update my status on the community forum. I will also contact Robotis for that Raspberry Pi hat you mentioned.

Thanks for all the time and effort you put into this issue!
Kind regards,
Benoît

Benoit-LdL, 2021-08-17:
I have a last question regarding the connection of the xl320's to the DXL port as you mentioned previously. In the OpenCm9.04 documentation [(4.3.12)](https://emanual.robotis.com/docs/en/parts/controller/opencm904/) it says the DXL port has following pinout:

D: direction control pin that controls TX and RX selection
X: TX pin
L: RX pin

While in the XL320 documentation [(3.1)](https://emanual.robotis.com/docs/en/dxl/x/xl320/) it mentions the following pinout:
1: GND
2: VDD
3: DATA

To me it looks like 2 different pinouts which can't be connected to each other... But maybe I missed something?

Kind regards,
Benoît

ROBOTIS-Will, 2021-08-18:
@Benoit-LdL 
Oh, the D/X/L pins are the UART pins + digital output pin that are directly connected to both MCU and the TTL IC.
[image]

The TTL IC(U8, U9, EF1) converts this TX and RX signal to 5V level DATA that can communicate with XL-320 while `D`(direction) controls the direction of data flow since DATA is a half duplex communication that uses a single line for transmitting and receiving data.

Benoit-LdL, 2021-08-18:
Hi @ROBOTIS-Will  thanks for the explanation! I will try it out.

Kind regards,
Benoît Lagasse

Benoit-LdL, 2022-05-06:
I've been able to control the XL320 servos directly from the Raspberry Pi using the OpenCM9.04 as a passthrough board.

I followed [this tutorial](https://community.robotis.us/t/rpi-driving-dynamixel-from-uart-at-1mbps/603).
In [this repository](https://github.com/Benoit-LdL/Robotis_Mini_Python) I created my own version (the main difference is that I'm using a Pi zero 2 and the resistor is on the RX line)