ROBOTIS-GIT/DynamixelSDK #511: [Proposal] Use a serial port with low latency mode on Linux

ssr-yuki, 2021-07-05:
Hi, thank you for maintaining.
Here I would like to propose an idea related to the following comments:

https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/c7e1eb71c911b87f7bdeda3c2c9e92276c2b4627/c%2B%2B/src/dynamixel_sdk/port_handler_linux.cpp#L33-L56

We can set the latency timer to 1 ms from C/C++ codes. 

Setting `ASYNC_LOW_LATENCY` flag to `serial_struct` struct used in [here](https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/c7e1eb71c911b87f7bdeda3c2c9e92276c2b4627/c%2B%2B/src/dynamixel_sdk/port_handler_linux.cpp#L245) is equivalent to setting the device latency timer to 1 ms for device communication. 

Once you agree with accelerating default communication speed of this SDK, I will implement it and send a PR soon.

ROBOTIS-Will, 2021-07-06:
Hi @ssr-yuki 

Thank you for your suggestion.

In recent DYNAMIXEL product such as X series, custom baud rates are not supported, but supported with limited number of popular baudrates.
https://emanual.robotis.com/docs/en/dxl/x/xm430-w210/#baud-rate8

However, I think it is a good practice to update the custom baud rate code for AX / MX series with protocol 1.0.
Is it also possible to implement a similar solution for the [setupPort()](https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/c7e1eb71c911b87f7bdeda3c2c9e92276c2b4627/c%2B%2B/src/dynamixel_sdk/port_handler_linux.cpp#L135) so that manual latency configuration is not necessary?

Thanks for your contribution!

ssr-yuki, 2021-07-06:
@ROBOTIS-Will 
Thank you for replying, and I am sorry for lack of my explanation.
This proposal is related to not custom baud rates but just communication speed.

The main purpose of this proposal is to reduce latency in packet reception.
In the current implementation, the value of latency timer is default, i.e. 16 ms. This is a clear bottleneck of serial communication.
To deal with this problem, two example methods are shown in the comments I refer to above, and it had been discussed in #116 .

**Using `ASYNC_LOW_LATENCY` enables the SDK to communicate with Dynamixel actuators 16 times faster (latency in packet reception becomes 16 ms -> 1 ms by default), without any additional operations**.

ROBOTIS-Will, 2021-07-06:
@ssr-yuki 
Oh, I'm sorry about the misunderstanding.
The LATENCY_TIMER in DYNAMIXEL SDK is used when status packet is expected from DYNAMIXEL and set the timeout error when the packet isn't received in time.
In most usual cases when packets are received normally, this latency won't cause a bottleneck of serial communication.
Decreasing the timeout latency may affect applications with lower baudrate such as 9600bps (which would be very rare though) I'm worried about decreasing the default value.
Please feel free to discuss further and if you can also provide cases that can reproduce such bottleneck, it would greatly help us to improve this latency issue.
Thanks!

ssr-yuki, 2021-07-06:
@ROBOTIS-Will 

> Decreasing the timeout latency may affect applications with lower baudrate such as 9600bps (which would be very rare though) I'm worried about decreasing the default value.

You are right, and I did not consider this case, however, I think it is not a critical problem. Timeout values should depend on not only the latency timer but also baud rates.

> you can also provide cases that can reproduce such bottleneck, it would greatly help us to improve this latency issue.

Thanks. The 16ms latency timer prevents us from accelerating control/sensing rates. The fastest loop rate with the default latency timer is 1 s / 16 ms =  62.5 Hz, although it can be up to 1000 Hz with the low latency mode. 

Currently this can be resolved by executing either of the methods written in the comments (I show it again):

https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/c7e1eb71c911b87f7bdeda3c2c9e92276c2b4627/c%2B%2B/src/dynamixel_sdk/port_handler_linux.cpp#L33-L56

and I insist that the SDK can execute the same operation. Changing `LATENCY_TIMER` macro variable is not an essential point here.

By this proposal I would just like to make the SDK performance better, independently from Linux changes.
I should ask you whether you want to use the low latency mode for better performance by default.

* Note that the latency timer value was 2 ms about four years ago. See [here](https://github.com/ROBOTIS-GIT/DynamixelSDK/commit/74d2bf6ca77dc78736bd8d81dbe8a7be05ce8789).

ROBOTIS-Will, 2021-07-07:
@ssr-yuki 
I think there's a misunderstanding about the LATENCY_TIMER.
The `LATENCY_TIMER` macro defined in the DYNAMIXEL SDK does NOT represent the latency of the serial port of the hardware device, but only used in the SDK to check if the packet is received in time so that the next instruction packets can be prepared.
https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/c%2B%2B/src/dynamixel_sdk/port_handler_linux.cpp#L161

I think the macro name and the comments next to it describes the serial port latency setting misleads what the LATENCY_TIMER actually does in the SDK.

ssr-yuki, 2021-07-07:
I have understood it. Again, Changing LATENCY_TIMER macro variable is not an essential point here.

I am just proposing to implement C/C++ codes not to require Method 1 nor Method 2.

ssr-yuki, 2021-07-07:
I will open a draft PR for this explanation. Please wait and review it.

ROBOTIS-Will, 2021-07-08:
@ssr-yuki 
Sounds good. I'll wait for your PR.
Thank you!

ssr-yuki, 2021-07-19:
@ROBOTIS-Will Sorry, I was busy last week. I open a PR showing what I suggest now.

ROBOTIS-Will, 2021-07-21:
Hi @ssr-yuki 
Thank you very much for the PR.
I'll review your PR, but please allow me some time.
Thanks!

adob, 2024-03-06:
What's the status here?