ROBOTIS-GIT/DynamixelSDK #512: XM430-W210 - motor velocity measurements

abb-9, 2021-07-07:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  - ex) `3.4.1`

I am not using a SDK. The issue is not related to the SDK.

2. Which programming language/tool do you use?
  - ex) `C`, `C++`, `C#`, `Python`, `Java`, `MATLAB`, `LABVIEW`, `ROS`, `Arduino`, etc

I use C++.

3. Which operating system do you use?
  - ex) `Ubuntu 16.04`, `Windows 10`, `OS X 10.14`

Custom code running on a microcontroller

3. Which USB serial converter do you use?
  - ex) `USB2Dynamixel`, `U2D2`, `OpenCM904`, `OpenCR`, etc

None of these

4. Which DYNAMIXEL do you use?
  - ex) `MX-28(2.0)`, `AX-12W`, `XM430-W250`, etc.

XM430-W210R

5. Have you searched the issue from the closed issue threads?

Yes

6. Please describe the issue in detail

The motor is operated in several different modes (position, velocity, PWM), and measurements corresponding to the position and velocity are logged. Numerical integration of the velocity measurements should give an approximate position with known initial position and velocity. However, the position obtained through numerical integration and the measured position obtained directly from the motor are in disagreement. To ensure that the numerical errors are low, the data from the motor is recorded at a sample rate of 1 kHz. The operating limits (velocity, current, PWM) of the motor and the profile have been suitable configured depending on the operating mode.

a. How is the motor velocity computed within the motor, because I believe the motor is only equipped with a position sensor?
b. What is the maximum request rate with which the motor can process commands?
c. What is the internal controller's update loop rate?

7. How can we reproduce the issue?

This issue can be reproduced by running the motor in any of the three modes, and logging both position and velocity measurement data. Performing a numerical integration of the velocity data to produce position given initial conditions of both the position and velocity will give you your computations. Compare the numerically integrated version and the version logged by the motor to note the discrepancy, which is the issue of this ticket.

ROBOTIS-Will, 2021-07-21:
Hi @abb-9 
Please see below answers to your inquiries.

a. The velocity is calculated based on the value read from the absolute position.
b&c. The max request rate would be depending on the loop rate of the internal controller, which can run faster than 1kHz. Please understand that we do not disclose a specific rate. In most cases, the control frequency is affected by other factors such as delays in communication interface(USB-serial latency introduced by the OS) 

Integration of discrete information inevitably introduces some errors. It is recommended to adopt other sensors that can compensate these errors if your application is critical to minor errors.
Thank you.

yun-goon, 2025-02-13:
It looks like the issue has been resolved, so I will close it now.

closed 2025-02-13 by yun-goon