ROBOTIS-GIT/DynamixelSDK #513: There is no status packet!

Grices, 2021-07-11:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  - ex) `3.7.51`

2. Which programming language/tool do you use?
  - ex) `Python`

3. Which operating system do you use?
  - ex) `Ubuntu 16.04`

3. Which USB serial converter do you use?
  - ex) `USB2Dynamixel`

4. Which DYNAMIXEL do you use?
  - ex) `MX-28(2.0)`

5. Have you searched the issue from the closed issue threads?
YES, but none of them worked.
6. Please describe the issue in detail
How do I ues TX2 drive motor and what are the steps? I have try install SDK as this web"https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/python_linux/#python-linux" but when I run any test code, there is"There is no status packet!"
7. How can we reproduce the issue?
After downloading the SDK3.7.51, cd "DynamixelSDK-3.7.51/python" run"sudo python setup.py install", then cd "/test/protocol1_0" run "python ping.py
[dynamixel.txt](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/6796891/dynamixel.txt)
"

ROBOTIS-Will, 2021-07-12:
Hi,

Please take a look at below quick start video.
https://youtu.be/LAizFTTdL8o
Thanks!

Grices, 2021-07-12:
Thank you very much!

Grices, 2021-07-12:
> 
> 
> Hi,
> 
> Please take a look at below quick start video.
> https://youtu.be/LAizFTTdL8o
> Thanks!
Thank you very much!

closed 2021-07-12 by Grices