ROBOTIS-GIT/DynamixelSDK #514: How to 'port_handler_ChibiOS'

Wilzzer, 2021-07-14:
1. Which DYNAMIXEL SDK version do you use?
  - 3.7.31

2. Which programming language/tool do you use?
  - C

3. Which operating system do you use?
  - ChibiOS 20.3.x

3. Which USB serial converter do you use?
  - ex) `USB2Dynamixel`, `U2D2`, `OpenCM904`, `OpenCR`, etc

4. Which DYNAMIXEL do you use?
  - XM430-W250

5. Have you searched the issue from the closed issue threads?
- Yes

6. Please describe the issue in detail
Hi !

We want to use our Dynamixel on a microcontroller controlled by a tablet running on ChibiOS 20.3.x. From the DynamixelSDK's documentation, it seems that we'd only need to implement a personnalized port_handler_chibios.c based on existing port_handler files. How would you proceed for this ? Would this already be implemented somewhere ? Do you have any advices ?

Thank you, in advance, for your answer!

I wish you a pleasant day.

ROBOTIS-Will, 2021-07-15:
Hi @Wilzzer 
Sorry, I do not have much experience with the RTOS so the information that I can provide might be very limited.
Depending on DYNAMIXEL model you are using (XM430-W250-T or XM430-W250-R), you might need an additional hardware to convert the UART signal into 5V level TTL or RS-485 and vice versa.
Just like you mentioned, you can create additional port handler file so that ChibiOS can access to a specific UART for transmitting and receiving packets between the tablet and DYNAMIXEL.
Thanks!

yun-goon, 2025-02-13:
Closing this thread as there isn't recent activity. Please feel free to reopen when necessary.

closed 2025-02-13 by yun-goon