ROBOTIS-GIT/DynamixelSDK #515: Dynamixel MX28(1.0)_Arduino Uno_Arduino IDE_ return to o position from unknown position when the power is turned on

Danaish-Rathore, 2021-07-16:
Hi everyone, I am new at Robotics. I am  controlling MX28 using Arduino Uno and working on the extended position control. 

Now the problem is when the power is off  and I rotate the MX28 servo manually suppose from  0 degree to 460 degree, in this case when switch up the power the dyanmixel should run way back to 0 position (which is set as home position) which it does. but it seems the servo moves from 460 to 360 only which is only 100 degrees back. 

I am working on a Project which needs up to 2 complete cycles in CCW and CW directions which is working fine when the dynamixel is connected to power. But once the torque is disable (i.e., power turned off) the motor can turn freely in any of the directions which causes problem as the potentiometer recognises the unknown position in limits between 0-360 but it doesn't know the number of cycles/turns when the power is turned again. 

Is there any way to solve this issue! using any program coding technique, as we know when the motor is connected to battery we easily know the number of turns/present extended position but when it is disconnected then we can only read the position in between 360 degrees no matter how many turns we rotate.

Thanks again.

ROBOTIS-Will, 2021-07-16:
Hi @Danaish-Rathore 

When configuring DYNAMIXEL as multi turn(or Extended position with Protocol 2.0 firmware), turn count is not saved in the EEPROM, therefore, after the power switch the Present Position will be reset to a value within 1 revolution (0 ~ 360 degree).
In order to recover the turn count after power reset, you should use your memory in the Arduino.

Thank you.

Danaish-Rathore, 2021-07-16:
Hi ROBOTIS-Will 

Thanks for prompt response.  It means I should save the data in EEPROM to recover the turn count and the problem will be solved, am I correct? 

I have one more question please is it possible to change the default velocity of the MX-28 permanently! 

Again thanks.

ROBOTIS-Will, 2021-07-16:
@Danaish-Rathore 
[MX-28 with Protocol 1.0](https://emanual.robotis.com/docs/en/dxl/mx/mx-28/) (factory default) is an old firmware with relatively less features compare to the Protocol 2.0.
[MX-28 (2.0)](https://emanual.robotis.com/docs/en/dxl/mx/mx-28-2/) can be burned via [DYNAMIXEL Wizard 2.0 firmware recovery feature](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#firmware-recovery).

With MX-28 (2.0) + Position Control mode, you can configure the maximum velocity with the Profile Velocity(112) as shown in the below, but this value will be reset after power switch.
[image]

However, in the case of MX-28 (2.0) + Velocity Control mode, you can set the velocity with [Goal Velocity(104)](https://emanual.robotis.com/docs/en/dxl/mx/mx-28-2/#goal-velocity104) and the Goal velocity cannot exceed the Velocity Limit(44) in the EEPROM which is saved in the non-volatile area of DYNAMIXEL.
[image]

Please note that the Arduino program written for Protocol 1.0 may not fully compatible with Protocol 2.0 DYNAMIXEL.

Thank you.

Danaish-Rathore, 2021-07-16:
**@ROBOTIS-Will ROBOTIS-Will**

Many thanks.

closed 2021-08-05 by Danaish-Rathore