ROBOTIS-GIT/DynamixelSDK #517: There is no status packet !

dkwtfid, 2021-07-26:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  No idea

2. Which programming language/tool do you use?
  I can't code :(

3. Which operating system do you use?
  Windows 10

3. Which USB serial converter do you use?
  OpenCR

4. Which DYNAMIXEL do you use?
   XM430-W210

5. Have you searched the issue from the closed issue threads?
yes

6. Please describe the issue in detail
I've been trying to set up waffle pi robot. Dynamixel won't move so i'm following this tutorial https://emanual.robotis.com/docs/en/platform/turtlebot3/opencr_setup/ and this one to set up Arduino https://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#setup-dynamixels-for-turtlebot3
I've basically been walking in the dark since the begining of the set up process and can't find any way to move forward since There is no status packet issue Arduino

7. How can we reproduce the issue?
I'm so lost pls help

ROBOTIS-Will, 2021-07-27:
Hi @dkwtfid 
Ok, let's see in which step you lost.
TurtleBot3 comes with preconfigured DYNAMIXEL module, but just to make sure that everything is setup correctly, I'll walk you through the DYNAMIXEL setup.
Let me know if you are lost in a specific step.

1. Install Arduino IDE on your Windows ([Follow this instruction](https://emanual.robotis.com/docs/en/parts/controller/opencr10/#install-on-windows))
2. Connect the micro USB to OpenCR and connect to the PC.
3. Run Arduino IDE and open examples from `File > Examples > TurtleBot3 > TurtleBot3_setup > turtlebot3_setup_motor`
  [image]

4. Connect the 12V SMP to OpenCR and turn on the power
5. Upload the Arduino example to OpenCR. In Windows, you'll see something like `COM ##` for the port instead of `dev/ttyACM0`.
  [image]

6. Open the terminal and connect ONLY ONE DYNAMIXEL at a time to setup. Repeat this step for left and right DYNAMIXEL.
  [image]

7. Once DYNAMIXEL setup is completed and tested, recover the TurtleBot3 firmware as instructed in the eManual ([step 10](https://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#setup-dynamixels-for-turtlebot3))

dkwtfid, 2021-07-27:
Thank you so much!

closed 2021-07-27 by dkwtfid