ROBOTIS-GIT/DynamixelSDK #521: [C++][STM32] Frame error on first group sync write

schorr31, 2021-08-09:
1. Which DYNAMIXEL SDK version do you use?
  - `3.7.40`

2. Which programming language/tool do you use?
  - `C++`

3. Which system do you use?
  - `FreeRTOS on STM32 with half duplex serial`

3. Which DYNAMIXEL do you use?
  - `XL430-W250 `

4. Have you searched the issue from the closed issue threads?
  - `yes`

5. Please describe the issue in detail
  - `The first group sync write after the initialisation produces a frame error in the header. The rest of the message is transmitted correctly. See the attached pictures for the first and the second message.`

[image]
[image]

ROBOTIS-Will, 2021-08-10:
Hi @schorr31 
Thank you for reporting the issue.
I'll check on this and get back to you.
Are you running on ROS2?

schorr31, 2021-08-10:
No i ported the C++ code to work with a STM32 microcontroller. It utilizes the ST hal library for hardware access. The program uses FreeRTOS, which is a real time operating system for microcontrollers.

ROBOTIS-Will, 2021-08-10:
@schorr31 
Sorry, but the dev environment isn't what I can test of.
Can you check the content of the txpacket array when the frame error occurs?
Right before the TX transmission, header is inserted as a byte on each field and skipping this is very unlikely.

schorr31, 2021-08-13:
Hello, the problem was the implementation of the STM32 porthandler. Thanks for your help!

closed 2021-08-13 by schorr31