ROBOTIS-GIT/DynamixelSDK #522: is SDK Compatible with OpenCM

syakirahmd, 2021-08-09:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  - `3.7.31`

2. Which programming language/tool do you use?
  - `Python`

3. Which operating system do you use?
  - `Windows 10`

3. Which USB serial converter do you use?
  - `OpenCM904`

4. Which DYNAMIXEL do you use?
  - `AX-18`

5. Have you searched the issue from the closed issue threads?
  - `yes` 

6. Please describe the issue in detail
`I am working on OpenCM 9.04 with python language using VScode. I am trying to move ax-18 with example code from SDK but its not working because some error '[TxRxResult] Incorrect status packet!'. So is SDK compatible with OpenCM 9.04? FYI I also try SDK using arduino IDE and it work well. I have to deal with python for my project. thanks`

ROBOTIS-Will, 2021-08-10:
Hi @syakirahmd 
Thank you for your inquiry.
DYNAMIXEL SDK unfortunately does not support python for OpenCM9.04.
OpenCM9.04 supports Arduino IDE via ported C++ DYNAMIXEL SDK.
Thank you.

syakirahmd, 2021-08-10:
Iam sorry could you tell me how to use it? or maybe there is a tutorial that you could share

syakirahmd, 2021-08-13:
thank a lot. I greatly appreciate you.

closed 2021-08-13 by syakirahmd