ROBOTIS-GIT/DynamixelSDK #523: Motor won't move under load

lara-z, 2021-08-11:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

**1. Which DYNAMIXEL SDK version do you use?**
  Protocol 2.0

**2. Which programming language/tool do you use?**
  Python

**3. Which operating system do you use?**
  Windows 10

**3. Which USB serial converter do you use?**
 U2D2

**4. Which DYNAMIXEL do you use?**
  XM 430-W210-T

**5. Have you searched the issue from the closed issue threads?**
  I did not find this issue elsewhere

**6. Please describe the issue in detail**
  I have been using my dynamixel under no load without issue. Recently, I connected to a tensioned, closed-loop cable system such that the tensioned cable kind of runs around the motor like a tractor tread so that even when the motor is off the system is under load. In these conditions, when commanded with python code, the motor started working (I heard it start to move) but was unable to move the desired amount. Surprisingly, there is no error registered in the Dynamixel Wizard 2.0. If I move the tensioned cable so that there is enough slack for the motor to move it its desired position, it does so and successfully completes the movement. I noticed, however, that with the same tensioned setup when I manually make the motor move in your dynamixel Wizard 2.0 interface using that wheel icon on the right hand side to rotate the motor, the motor does actually move the specified amount without any manual assistance. So it can move under load.
  
  What type of motor control does the Dynamixel Wizard interface use so that it is able to move the motor? Does it changed to current-based control so that it can increase the current or voltage that the motor receives (to increase the torque)?

I tried to control the motor with current-based position control to see if I could achieve a higher torque. I’m still puzzled by the fact that I could get the motor to move properly when using your Dynamixel Wizard position dial but not with my code. The motor again stalled, even when I specified a goal current of 3A and pwm of 100%.

When doing current-based position control, I read out the current, PWM, and voltage from the motor. To make sure I had successfully specified the current/PWM/voltage limits and goals, I read those out from the motor as well. They seemed fine. Below is an image from when the motor stalled where the current reading has been converted to amps and the voltage reading to volts based on the motor spec sheet. The pwm I left in the original units coming straight from the motor. I was actually surprised that the current and pwm were so low compared to their goal values.

[image]
[code.zip](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/6965096/code.zip)

**7. How can we reproduce the issue?**
I am attaching the code. If you connect a run a tensioned cable around the motor and run my code, it should reproduce the issue. Please run test.py which references utils.py.  The code never exits the while loop of the move() function in line 229 of utils.py.

ROBOTIS-Will, 2021-08-12:
Hi @lara-z 
It seems like you are giving a very small range of Goal Position for DYNAMIXEL.
Please note that under the single turn (Position Control Mode), DYNAMIXEL only rotates within a 0 ~ 359 degree and never crosses the 0 and 360 border. I've modified the goal_position in line 11 of test.py as below and confirmed that it works from 1048 ~ 3048.
```python
goal_position = [2048 + direction*1000]
```
Please be aware that this range may not acceptable to your application so please test with a separate module that is not connected to your tensioned cable.
When I tested with my setup, I was grasping the wheel connected to DYNAMIXEL but it worked fine.

lara-z, 2021-08-13:
Hi Will,

Thank you for looking into this! Unfortunately, the solution you proposed doesn't seem to solve the problem. When I increased the magnitude of the desired motion, the motor still doesn't want to move. However, moving it with the Dynamixel Wizard application still works. I've been using extended position control (as opposed to position control mode), so I think hitting the position limits isn't an issue. It seems to me that it's the current/torque that the motor exerts that's the issue here since the position range isn't limited and the current and PWM readings and seem extremely low?

ROBOTIS-Will, 2021-08-17:
Hi @lara-z 
Please see the attached video that I ran your code without modification (except the COM_PORT)
This is tested with factory recovered XM430-W210 with Extended Position Control Mode, 1Mbps Baudrate configuration.

https://user-images.githubusercontent.com/26239359/129663199-1323ec5f-8de9-4c08-93cf-5946c5298f72.MOV

yun-goon, 2025-02-13:
Closing this thread as there isn't recent activity. Please feel free to reopen when necessary.

closed 2025-02-13 by yun-goon