ROBOTIS-GIT/DynamixelSDK #525: Can't import ‘dynamixel-sdk’, on python and Matlab

SamCoTuck, 2021-08-28:
I am getting an error when I tried to run Read-Write.py

dynamixel_sdk is already installed, 

Python error message:
ModuleNotFoundError: No module named 'dynamixel_sdk' 

Matlab error message:
Error in read-write line 48

![image](https://user-images.githubusercontent.com/68170403/131208030-5f7d1bdb-5f9f-4a8a-bfc4-ff1bc197a58e.png

[image]

ROBOTIS-Will, 2021-08-30:
Hi @SamCoTuck 
Have you setup the DYNAMIXEL SDK library as well?
https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/python_windows/#building-the-library

SamCoTuck, 2021-08-30:
@ROBOTIS-Will I am using Mac OC. Thanks

ROBOTIS-Will, 2021-09-06:
Hi @SamCoTuck 
I'm sorry about the delayed response.
I believe the Mac environment isn't much different from Linux.
Have you properly imported all necessary libraries in the Matlab?
https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/library_setup/matlab_linux/#matlab-linux

yun-goon, 2025-02-13:
Closing this thread as there isn't recent activity. Please feel free to reopen when necessary.

closed 2025-02-13 by yun-goon