ROBOTIS-GIT/DynamixelSDK #526: Unable to detect U2D2 USB port

jalpanchal, 2021-09-06:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.
Issue :

Hi,

I have 4 Dynamixel XC430-W24Ts with a U2D2 and a power hub. 
I have been using these for over 6 months now with no issues. 
But now when I make the  USB connections to a computer, the U2D2 port is not identified. I dont see it in the port list of the OS or in the Wizard 2.0 software.

I have tried using different cables and unplugging and reconnecting everything. I use a  12 V power supply and I checked with a multimeter that it is providing the right power.

When I connect the power supply, I do see a red light flash on the motor telling me that it's probably okay, and when the USB is connected, the red power light on the U2D2 also goes on. 

I have tried this with a Mac and a Windows machine and I see the same behavior that the U2D2 USB port is not visible in the OS or the Wizard software.

Please do advise on how do I debug this issue.

Thank you.

Jal

Details : 
1. Which DYNAMIXEL SDK version do you use?
  -  3.7.51

2. Which programming language/tool do you use?
  - MATLAB and Python

3. Which operating system do you use?
  - OS 11.5.2/ Windows 10

3. Which USB serial converter do you use?
  - U2D2

4. Which DYNAMIXEL do you use?
  - XC430-W240T

ROBOTIS-Will, 2021-09-06:
Hi @jalpanchal 

The U2D2 does not share the power from connected DYNAMIXEL, but only supplied via USB.
If the device is not detected from multiple different PCs and different OSes, the product may have been damaged.
Please also check if the USB port of your PC is working with other USB devices.
Thanks.

jalpanchal, 2021-09-06:
Thank you for the update. I did check that the USB port is working. It's strange that the device stopped working randomly.

ROBOTIS-Will, 2021-09-07:
Yes, it is strange, but many things happen without clear reasons like an engine check warning on my car yesterday.
I guess it is just a bad luck.

yun-goon, 2025-02-13:
Closing this thread as there isn't recent activity. Please feel free to reopen when necessary.

closed 2025-02-13 by yun-goon