ROBOTIS-GIT/DynamixelSDK #527: Fast Sync Read and Fast Bulk Read support

TJMolnar, 2021-09-20:
Hi,

I see in the documentation for protocol 2.0 on the e-manual (https://emanual.robotis.com/docs/en/dxl/protocol2/) definitions for Fast Sync Read and Fast Bulk Read methods, is there a timeline for supporting these in the SDK?

Thanks

ROBOTIS-Will, 2021-09-28:
Hi @TJMolnar 
Thanks for the inquiry.
The Fast Sync / Bulk will be supported in October.

TJMolnar, 2021-10-05:
@ROBOTIS-Will Thanks for the update!

gnastacast, 2022-04-18:
Any movement on this? As far as I can tell it is still unsupported. I was thinking about taking a crack at it this week.

ROBOTIS-Will, 2022-04-19:
Hi @gnastacast 
Sorry about the delay.
At the moment, you may try the Fast Sync Read in the `develop` branch.
https://github.com/ROBOTIS-GIT/DynamixelSDK/commit/d57e57d78fe002056fd582741035f354c6845342
New APIs regarding the Fast Read functions will be revised in the first half of this year.
Thank you.

CImogu, 2022-07-27:
Hello. I was wondering about the status of the new APIs regarding the Fast Read functions. I tried looking but I could not find any documentation.

yun-goon, 2025-03-14:
Hello,

Please check the following video:
🔗 [YouTube Link](https://www.youtube.com/watch?v=claLIK8omIQ)

If you encounter any issues while using it, please let us know.

closed 2025-06-20 by sunghowoo