ROBOTIS-GIT/DynamixelSDK #529: Support to read multiple data addresses for multiple dynamixels

mrunaljsarvaiya, 2021-09-23:
The cpp interface has support to read a range of data for multiple dynamixels i.e A single call to `syncRead` can give me data for position,velocity and current for two dynamixels. Is there support for a similar function in the python interface? Seems like the python `syncRead` can only be used to read a specific address for multiple dynamixels

ROBOTIS-Will, 2021-09-27:
Hi @mrunaljsarvaiya 

The SyncRead in the Python works the same as that of the CPP.
In order to read multiple sequential addresses, you can initialize the sync read instance with the beginning address and the total length of the data to read.

https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/python/tests/protocol2_0/sync_read_write.py#L126

For example, initializing the above with `LEN_PRESENT_POSITION = 8` will read 8 byte from the beginning of the Present Position.
Getting data in the below can be iterated with a proper parameter until reading both Present Position and the Velocity Trajectory.

https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/python/tests/protocol2_0/sync_read_write.py#L225

mrunaljsarvaiya, 2021-09-27:
Got it thanks! 

Does there happen to be python support to convert the present position to a signed decimal value? When I read the present position using `GroupSyncRead` I get back a very high number if the position is negative (confirmed with the dxl GUI)

ROBOTIS-Will, 2021-09-28:
For the negative values, we take 2's complement and this is why getting a very large value when the negative value is expected.
https://github.com/ROBOTIS-GIT/DynamixelSDK/issues/264

mrunaljsarvaiya, 2021-09-30:
> For the negative values, we take 2's complement and this is why getting a very large value when the negative value is expected. Please refer to the thread #264

Got it, thanks!

closed 2021-09-30 by mrunaljsarvaiya