ROBOTIS-GIT/DynamixelSDK #530: Matlab reported that it could not find the file dynamixel_sdk.h.

w-1016-Papman, 2021-09-29:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  The version I use is DynamixelSDK-3.7.31

2. Which programming language/tool do you use?
  MATLAB 2021A

3. Which operating system do you use?
  Windows 10

3. Which USB serial converter do you use?
  USB2Dynamixel

4. Which DYNAMIXEL do you use?
XM430-W250

5. Have you searched the issue from the closed issue threads?

6. Please describe the issue in detail
Wrong use loadlibrary cannot find the file dynamixel_sdk.h. Error loadlibrary error read_write (line 48) [notfound, warnings] = loadlibrary(lib_name,'dynamixel_sdk.h','addheader','port_handler.h','addheader','packet_handler.h');
7. How can we reproduce the issue?

w-1016-Papman, 2022-01-15:
> Hi @w-1016-Papman I'm sorry about the delayed response. Please see below video instruction that describes how to use DYNAMIXEL SDK with Matlab. https://youtu.be/1VWksgPqyDs Thanks!

I am very grateful for your reply and attention to the question. As the New Year, the most important festival in China, is approaching, I sincerely wish you all the best.

ROBOTIS-Will, 2022-01-17:
@w-1016-Papman 

Thanks for the feedback.
Happy New Year and wish the best for you too!

yun-goon, 2025-02-14:
I think the problem has been solved, so I will close this issue.

closed 2025-02-14 by yun-goon