ROBOTIS-GIT/DynamixelSDK #532: [3.7.51][Python3][U2D2][XM430-W350-T] Issue : read4ByteTxRx in protocol2_packet_handler.py return empty data ?

ZhengYi0310, 2021-10-20:
Hello,
    I am using the SDK on ubuntu 20.04, with other specifications listed in the title, here is the code I have:
  
[image]
[image]
[image]
[image]

When I run the script, it seems the communication stops randomly after the script starts running, sometimes less than 1 minute, sometimes about 5 to 10 minutes, and the error is the following:

it seems the readTxRx function receives empty data. Also, the LED changes to static red, another indicator that the communication stops.

[image]
I've already changed to the baud rate (default is 57600) to a lower and higher number, but it seems not relevant.

Thanks in advance !

ZhengYi0310, 2021-10-31:
Hi  @ROBOTIS-Will 
     More updates, when I run the read_write.py script in DynamixelSDK/python/tests/protocol2_0/ the same issue happens, as can be seen in the following
[image]

    But when I use Dynamixel Wizards 2, the issue never happens, hence I don't think it's a cable issue, and I suspect it is related with the read and write frequency.

    Thanks
:

ROBOTIS-Will, 2021-11-03:
Hi @ZhengYi0310 
If you are getting similar issues with the example code that we provide, I think your environment may cause unstable communication.
Since you are getting an intermittent "no status packet" error, I'd check your communication cables.
If possible, try testing with a shorter USB cables as extended USB cables that are longer than 1m(3 feet) could affect stable communication.
Thank you.

yun-goon, 2025-02-14:
I think the problem has been solved, so I will close this issue.

closed 2025-02-14 by yun-goon