ROBOTIS-GIT/DynamixelSDK #537: Real time motion in dynamixel motors.

Raj-Kiran127, 2021-11-25:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  - ex) 3.7.31

2. Which programming language/tool do you use?
  - ex) MATLAB

3. Which operating system do you use?
  - ex)Windows 10

3. Which USB serial converter do you use?
  - ex) U2D2

4. Which DYNAMIXEL do you use?
  - ex) `MX-28 AT (protocol 1.0)
  
5. Have you searched the issue from the closed issue threads?
   Yes

6. Please describe the issue in detail
   I need to run 3 motors in real time. But as I increase the number of steps motors are taking more time and the motion is not smooth. So I used sync_write, it worked to some extent but am not getting smooth motion and also motors are not completing the motion in given time. How can I trouble shoot the issue please let me know. 
An add up question - Are there any libraries of sync_write or bulk_read for simulink ?  
Thanks in advance.
7. How can we reproduce the issue?
[TrialWithSyncWrite.txt](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/7604269/TrialWithSyncWrite.txt)

This is the code am trying to run. Please bear with my improper coding style.

ROBOTIS-David, 2021-12-02:
Hi Raj, 

Your issue has been addressed via Tech Support, 

See the quote below, 
_1) Giving position control(in matlab), when I try to increase the number of steps, motors slow down. How can I achieve a smooth motion for a given position control without slowing down the motors? (code - To_Fro.m)_
- Seems that you are updating Goal Position for three DYNAMIXEL at a very detailed time within a loop. The problem in this case, you might see motion jittering or regulated motion speed depending on Baudrate or any control period. 
I recommend updating position trajectory before the current position reaches the next Goal Position, so that you may see the movement without jittering (See below example). And I highly recommend using a higher baudrate to transmit more command packets at a given time.

i.e)
- Final Position: 2048
- Moving: 1030 -> 1080 (When Present Position is reached before 1050, Goal Position Position should be updated the next)-> 1050 -> 1060 ..... 2048. 

_2) Using simulink library for dynamixel motors available on matlab file exchange I achieved a smooth motion for 1 motor which finished in the given time but when I try to run all the motors at once, motion is  not smooth and the end time is increasing. 
- To be honest, I am not used to Matlab (and we do not give simulink tech support for third party libraries), therefore this is difficult for me to give you relevant assistance.
- If you see motion discrepancy, use Sync Write Instruction rather than basic Read/Write. 
_3) How can I run the motors with real time using position control?_ 
- It's vague for me to understand the point of it. Please give me more clarification on the real-time control in Joint Mode. There are many different concept in real-time (Such as building a loop controller out of DYNAMIXEL)

yun-goon, 2025-02-13:
Closing this thread as there isn't recent activity. Please feel free to reopen when necessary.

closed 2025-02-13 by yun-goon