ROBOTIS-GIT/DynamixelSDK #538: Ping example - [TxRxResult] There is no status packet!

t-furst-kinetic, 2021-12-01:
ISSUE TEMPLATE ver. 1.2.0

1. Which DYNAMIXEL SDK version do you use?

    - 3.7.51

2. Which programming language/tool do you use?

    - C++

3. Which operating system do you use?
    - Linux x64: ZorinOS (based on Ubuntu 20.04.3 LTS)

3. Which USB serial converter do you use?

    - OpenCR

4. Which DYNAMIXEL do you use?
    
    - XH540-W270T

5. Have you searched the issue from the closed issue threads?

    - Yes, but I didn't find any clear solution to my issue. Issues were closed with providing help.

6. Please describe the issue in detail

    - I'm trying to control a Dynamixel motor XH540-W270T with an OpenCR board and using the DynamixelSDK for the software part.
    I managed to communicate very well with the motor when using ArduinoIDE with the ping and scan examples. The motor even moved as expected using the position example.
  
    But when I try to do the exact same thing with the C++ examples in DynamixelSDK, I always get the following output:
  
    ```
    Succeeded to open the port!
    Succeeded to change the baudrate!
    [TxRxResult] There is no status packet!
    ```

    More information:
      - I use Protocol 2.
      - The motor is connected with TTL and is properly powered. I made sure that it has the correct ID.
      - Everything went well during the build of the SDK libraries.
      - I also tried the examples with Python, but I have the same result.

Do you have any idea of what could go wrong in my case?

Thanks.

ROBOTIS-Will, 2021-12-02:
Hi @t-furst-kinetic 

The OpenCR is not an official USB-serial converter like U2D2.
You may use Arduino sketch to run DYNAMIXEL directly from the OpenCR, but in order to use the OpenCR as a communication interface between DYNAMIXEL and PC, you should upload the `usb_to_dxl` Arduino sketch.
This sketch will relay packets between DYNAMIXEL and USB ports with some limitations.

Thanks!

t-furst-kinetic, 2021-12-02:
Hi @ROBOTIS-Will 

Thank you very much. That was the info I was missing.

closed 2021-12-02 by t-furst-kinetic