ROBOTIS-GIT/DynamixelSDK #539: overload and can't move to the goal position

chenxinencode, 2021-12-02:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  - ex) `3.4.1`
AX-12a
2. Which programming language/tool do you use?
  - ex) `C`, `C++`, `C#`, `Python`, `Java`, `MATLAB`, `LABVIEW`, `ROS`, `Arduino`, etc
matlab
3. Which operating system do you use?
  - ex) `Ubuntu 16.04`, `Windows 10`, `OS X 10.14`
`Windows 10
3. Which USB serial converter do you use?
  - ex) `USB2Dynamixel`, `U2D2`, `OpenCM904`, `OpenCR`, etc
rs232 to ttl
4. Which DYNAMIXEL do you use?
  - ex) `MX-28(2.0)`, `AX-12W`, `XM430-W250`, etc.
AX-12A
5. Have you searched the issue from the closed issue threads?

6. Please describe the issue in detail
I can't move the servo to the goal position via matlab,So I open the software of  Wizard ,It turns out the servo is overload,How can I fix the problem?
7. How can we reproduce the issue?

ROBOTIS-Will, 2021-12-03:
Hi @chenxinencode 

In case of overload error without any load, the DYNAMIXEL could be damaged.
Please perform [self diagnosis from Wizard 2.0](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#self-diagnosis)
Thanks

yun-goon, 2025-02-13:
It looks like the issue has been resolved, so I will close it now.

closed 2025-02-13 by yun-goon