ROBOTIS-GIT/DynamixelSDK #541: Can CM530 can be used to control dynamixel by coding in c program ,if yes the how?

ragav1111, 2021-12-04:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  -  `3.7.31`

2. Which programming language/tool do you use?
  -  `MATLAB`

3. Which operating system do you use?
  -  `Windows 10`, 

3. Which USB serial converter do you use?
  - ex) CM530 can be used instead of USB2Dynamixel

4. Which DYNAMIXEL do you use?
  - `AX-12 A

5. Have you searched the issue from the closed issue threads?

6. Please describe the issue in detail

7. How can we reproduce the issue?

ROBOTIS-Will, 2022-02-15:
Hi @ragav1111 

I'm sorry about the delayed response to your inquiry.
[The embedded C library for CM-530](https://emanual.robotis.com/docs/en/software/embedded_sdk/embedded_c_cm530/) will give you some freedom to program a custom firmware, but we do not provide a solution for specific custom feature.
Thank you.

ragav1111, 2022-02-15:
ok
i have a doubht , why we cannot use a function with sin as input to dynamixel  write? while compelling  its show error

thanks for reply

[Amrita University]

ROBOTIS-Will, 2022-02-16:
@ragav1111 
I'm not quite sure what you are trying to do, but if you need specific math functions that are not included in the library, you'll need to include them.
Thanks.

closed 2025-02-04 by yun-goon