ROBOTIS-GIT/DynamixelSDK #543: buffer overrun warning and a heap corruption error

ROBOTIS-Will, 2021-12-22:
Hello everyone,
I am using C++ for running MX-28AT(2.0) motors with a U2D2 controller. I am using position control mode to run motors. When am dynamically allocating size for an array while using sync read- write am getting buffer overrun warning and group sync is giving me a heap corruption error. I am a beginner in coding, Can you please tell me how can I overcome this issue.

_Originally posted by @Raj-Kiran127 in https://github.com/ROBOTIS-GIT/DynamixelSDK/issues/258#issuecomment-998937697_

ROBOTIS-Will, 2021-12-22:
If you are getting such warnings and error with our default example, please let me know.
If you have modified the example and getting such warning and error, your program may not properly allocating the memory or accessing to the allocated memory.
In this case, you will need to get some insights from C++ communities such as stackoverflow by searching for similar cases.
Unfortunately, with a given information, it is difficult for me to identify the problem in our DynamixelSDK library.
Thanks.

Raj-Kiran127, 2021-12-24:
Hello Will,
Thanks for replying and moving the thread.
Ya I have tweaked the code of sync-write for my needs. But I was just trying to allocated memory dynamically in the code. I am attaching the pictures of the same. 
I have another issue as well, I was looking for faster communication and so set the baud rate to 4500000 using wizard 2.0 , I was able to run at that baud rate using Matlab but with C++ it is not working, am just getting the debugging message of code exited with 0. What could be the issue?
Also a follow up question - when I was using sync-read and write at the same time with n intermediate positions from say 100-4000 the motor were slowing down drastically. Is it because the half duplex communication where were writing and reading data at the same time or is it something else?
Please let me know, thank you.
[image]
[image]

ROBOTIS-Will, 2021-12-27:
Hi @Raj-Kiran127 

Thanks for attaching the picture, but I'm afraid that I can't help you with debugging your code.
In order to check if the problem is caused by the SDK library, please reproduce the issue with the default example in the SDK.
For example, if the 4.5Mbps doesn't work in C++, please modify the baudrate value in the C++ example in the SDK and share the result with us.

In case you need a maximum responsiveness from DYNAMIXEL, please set as below.
- Set the USB port latency of the PC to 1ms
- Set the [Return Delay Time](https://emanual.robotis.com/docs/en/dxl/mx/mx-28-2/#return-delay-time9) of DYNAMIXEL to 0

Please note that your Windows OS does not guarantee the exact timing control for sending and receiving packets.
In order to check the accurate timing between the Tx and Rx, please use an oscilloscope or a packet analyzer on data line of DYNAMIXEL.

Thanks.

Raj-Kiran127, 2022-01-22:
Hello @ROBOTIS-Will 
I have tried with the default code and they are working fine. But the issue is I have to give different positions to different motors. How can I do that using the default code. Some insight will be of great help.

ROBOTIS-Will, 2022-01-24:
Hi @Raj-Kiran127 

The Goal Position in the default Sync Read Write example is defined in the code below.
If you need to use other goal positions, you may revise the values or add new values.
https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/c%2B%2B/example/protocol2.0/sync_read_write/sync_read_write.cpp#L138

The example is written for P series with ID 1 and 2.
If you need to use other DYNAMIXEL such as X series or MX(with 2.0 firmware), you need to modify the control table definitions below.
```cpp
// Control table address
#define ADDR_PRO_TORQUE_ENABLE          562                 // Control table address is different in Dynamixel model
#define ADDR_PRO_GOAL_POSITION          596
#define ADDR_PRO_PRESENT_POSITION       611
...
#define DXL_MINIMUM_POSITION_VALUE      -150000             // Dynamixel will rotate between this value
#define DXL_MAXIMUM_POSITION_VALUE      150000
```

To add more DYNAMIXEL, you may append additional IDs in the code and add them as new parameter.
For example, to append ID3,

```cpp
#define DXL1_ID                         1                   // Dynamixel#1 ID: 1
#define DXL2_ID                         2                   // Dynamixel#2 ID: 2
#define DXL3_ID                         3                   // Dynamixel#3 ID: 3
...

  // Enable Dynamixel#2 Torque
  dxl_comm_result = packetHandler->write1ByteTxRx(portHandler, DXL2_ID, ADDR_PRO_TORQUE_ENABLE, TORQUE_ENABLE, &dxl_error);
  if (dxl_comm_result != COMM_SUCCESS)
  {
    printf("%s\n", packetHandler->getTxRxResult(dxl_comm_result));
  }
  else if (dxl_error != 0)
  {
    printf("%s\n", packetHandler->getRxPacketError(dxl_error));
  }
  else
  {
    printf("Dynamixel#%d has been successfully connected \n", DXL2_ID);
  }

  // Enable Dynamixel#3 Torque
  dxl_comm_result = packetHandler->write1ByteTxRx(portHandler, DXL3_ID, ADDR_PRO_TORQUE_ENABLE, TORQUE_ENABLE, &dxl_error);
  if (dxl_comm_result != COMM_SUCCESS)
  {
    printf("%s\n", packetHandler->getTxRxResult(dxl_comm_result));
  }
  else if (dxl_error != 0)
  {
    printf("%s\n", packetHandler->getRxPacketError(dxl_error));
  }
  else
  {
    printf("Dynamixel#%d has been successfully connected \n", DXL3_ID);
  }
...

  // Add parameter storage for Dynamixel#2 present position value
  dxl_addparam_result = groupSyncRead.addParam(DXL2_ID);
  if (dxl_addparam_result != true)
  {
    fprintf(stderr, "[ID:%03d] groupSyncRead addparam failed", DXL2_ID);
    return 0;
  }

  // Add parameter storage for Dynamixel#3 present position value
  dxl_addparam_result = groupSyncRead.addParam(DXL3_ID);
  if (dxl_addparam_result != true)
  {
    fprintf(stderr, "[ID:%03d] groupSyncRead addparam failed", DXL3_ID);
    return 0;
  }

...
    // Add Dynamixel#2 goal position value to the Syncwrite parameter storage
    dxl_addparam_result = groupSyncWrite.addParam(DXL2_ID, param_goal_position);
    if (dxl_addparam_result != true)
    {
      fprintf(stderr, "[ID:%03d] groupSyncWrite addparam failed", DXL2_ID);
      return 0;
    }

    // Add Dynamixel#3 goal position value to the Syncwrite parameter storage
    dxl_addparam_result = groupSyncWrite.addParam(DXL3_ID, param_goal_position);
    if (dxl_addparam_result != true)
    {
      fprintf(stderr, "[ID:%03d] groupSyncWrite addparam failed", DXL3_ID);
      return 0;
    }
...
etc...
```

Raj-Kiran127, 2022-01-26:
Hello @ROBOTIS-Will ,
Thanks for mentioning out the code for me. Lets say if I want to give goal positions which are of type float. How can I do the conversion of float to uint as the information is lost during the conversion. Can you please elaborate on how can I give goal positions of type float.

ROBOTIS-Will, 2022-02-03:
@Raj-Kiran127 
DYNAMIXEL only receives an integer value for its position.
If your function returns a float value with a decimal point, you'll need to convert this into an integer that maps the Goal Position of DYNAMIXEL.

Since 1 revolution maps into 0 ~ 4095, `0.08789 &deg; = 1 encoder tick`
https://emanual.robotis.com/docs/en/dxl/mx/mx-28-2/#goal-position116

Thanks.

yun-goon, 2025-02-13:
It looks like the issue has been resolved, so I will close it now.

closed 2025-02-13 by yun-goon