ROBOTIS-GIT/DynamixelSDK #545: extern variables in .h : linking error since gcc-10

yohanlediraison, 2022-01-21:
Hi,

**System :** Linux Debian 64 bits - but it's a problem linked to gcc & g++

**Problem:** several variables are not explicity declared "extern" in several .h files and cause a linking error since gcc 10
- **port_handler.h**, line 47 & 48 : g_used_port_num & *g_is_using
- **packet_handler.h**, line 82 :  PacketData *packetData;

the error is then, in the linking part (sorry for the french):
```
g++-10 -shared -fPIC -m64 -o ./libdxl_x64_c.so ./.objects/group_bulk_read.o ./.objects/group_bulk_write.o ./.objects/group_sync_read.o ./.objects/group_sync_write.o ./.objects/packet_handler.o ./.objects/port_handler.o ./.objects/protocol1_packet_handler.o ./.objects/protocol2_packet_handler.o ./.objects/port_handler_linux.o -lrt

/usr/bin/ld : ./.objects/group_bulk_write.o:/tmp/DynamixelSDK-3.7.31/c/build/linux64/../../include/dynamixel_sdk/port_handler.h:48 : définitions multiples de « g_is_using »; ./.objects/group_bulk_read.o:/tmp/DynamixelSDK-3.7.31/c/build/linux64/../../include/dynamixel_sdk/port_handler.h:48 : défini pour la première fois ici
```
with gcc-9, no error, since it's ignoring this error.
The problem and the solution is described here, in the 1st part:
https://gcc.gnu.org/gcc-10/porting_to.html

Just have to declare this variables external.
in **port_handler.h**
```
47  extern int     g_used_port_num;
48  extern uint8_t    *g_is_using;
```

in **packet_handler.h**:
`82  extern PacketData *packetData;
`

ROBOTIS-Will, 2022-01-24:
Hi @yohanlediraison 
Thanks for reporting the issue.
This change will be applied in the next release.

yohanlediraison, 2022-01-24:
Hi,

with this fix, i have problem to compil the examples and my own programs now. GCC complains in the linking part:
/usr/bin/ld : /usr/local/lib/libdxl_x64_c.so : référence indéfinie vers « g_used_port_num »
/usr/bin/ld : /usr/local/lib/libdxl_x64_c.so : référence indéfinie vers « packetData »
/usr/bin/ld : /usr/local/lib/libdxl_x64_c.so : référence indéfinie vers « g_is_using »

i'll investigate.

silver2row, 2022-06-07:
Hello,

I am facing a similar issue w/ a Linux SBC, 32-bit w/ compiling for the C language. Python3 and C++ compile just fine.

I started an updated issue in this repo. in case others are having similar concerns. 

Seth

P.S. Has the new release ever come to be?

a-gavin, 2023-01-13:
[This commit](https://github.com/ROBOTIS-GIT/DynamixelSDK/commit/91f6c43b27013ba844d6be0e64190f7c1389ebb0) doesn't completely fix the issue at hand. Declaring the variable as `extern` suppresses the issue at compile time, but the issue pops up at run time instead. The patch I just submitted addresses this.

To elaborate, declaring the variables as `extern` promises the linker that the code will define (and thus create storage) for the `extern`'ed variables elsewhere. However, the original fix does not define the variables elsewhere.

The core issue that causes this is that the `#ifndef` guards in the include files don't work as intended. I don't know the full reason why (something having to do with translation units?). That said, putting a `#warning "here"` below the erroneous globals (within the code protected by the `#ifndef`s) shows that these files are included multiple times and thus the variables are defined multiple times.

yun-goon, 2025-02-25:
Hello, I apologize for the late response.
This issue has been addressed and updated in the main branch through [Pull Request #641](https://github.com/ROBOTIS-GIT/DynamixelSDK/pull/641).
If you encounter any other issues, please feel free to let us know.
Thank you!

closed 2025-03-10 by yun-goon