ROBOTIS-GIT/DynamixelSDK #546: Typo in Matlab code

hucik14, 2022-01-22:
1. Which DYNAMIXEL SDK version do you use?
  - the latest master branch

2. Which programming language/tool do you use?
  - `MATLAB`

3. Which operating system do you use?
  - `Windows 10`

3. Which USB serial converter do you use?
  - `U2D2`

4. Which DYNAMIXEL do you use?
  - `XH430-270-R`

5. Have you searched the issue from the closed issue threads?
- no

6. Please describe the issue in detail
- file bulk_read_write.m shall have on line 183 parameter ID "DXL2_ID", not "DXL1_ID", ref:
https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/c7e1eb71c911b87f7bdeda3c2c9e92276c2b4627/matlab/protocol2.0/bulk_read_write.m#L183

7. How can we reproduce the issue?
- it's typo

ROBOTIS-Will, 2022-01-24:
Hi @hucik14 
Thanks for reporting the typo!
This change will be included in the next release.

yun-goon, 2025-02-13:
It looks like the issue has been resolved, so I will close it now.

commit a539a271 (2022-01-24): fix #546
closed 2025-02-13 by yun-goon