ROBOTIS-GIT/DynamixelSDK #548: Sync Write don't move motors

WilliamH07, 2022-02-11:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  - 

2. Which programming language/tool do you use?
  -Python, Raspberry

3. Which operating system do you use?
  - Raspberry OS lite

3. Which USB serial converter do you use?
  - U2D2 with power hub board

4. Which DYNAMIXEL do you use?
  - XL_320 and XL_430

5. Have you searched the issue from the closed issue threads?
   -Yes
6. Please describe the issue in detail

Hello, I use the DynamixelSDK in a raspberry pi 3 b+ to control my dynamixel using python. But when I run the code of example, everything works perfectly excepting the motor, they aren't moving but I can see on the serial their position. I try multiple Dynamixel like XL_430 and 320 but I don't works on them also. I don't understand because I did everything on the YouTube Tutorial. Thanks 

7. How can we reproduce the issue?

WilliamH07, 2022-02-11:
And I have also a another question, I use 2 XL_320 and 2 XL_430, can I use them at the same time ?

ROBOTIS-Will, 2022-02-15:
Hi @WilliamH07 

Which SDK example did you run for DYNAMIXEL?
Please note that although XL430 and XL-320 both use DYNAMIXEL Protocol 2.0, they have different Control Table structures and you should take these differences in consideration when running the example.

Also, please be aware that the operating voltage of XL430(11.1V) and XL-320(7.4V) are different.

WilliamH07, 2022-02-15:
Hi, I use the Dynamixel SDK 3.7.51
All of the Dynamixel are in daisy channel, for the voltage I use a voltage step-down to convert 12V to 7.4V for the XL320
I also try a XL430 alone, but like the daisy channel it didn't work too  
[image]
This is a screenshoot of my terminal 
Thanks

ROBOTIS-Will, 2022-02-17:
Hi @WilliamH07 

I see that the ID for the DYNAMIXEL in the code is set to `10`.
Please make sure that the DYNAMIXEL module with ID 10 (default : 1) is configured correctly with the baudrate (default : 57600) and the operating mode(default : Position mode).

You can connect the U2D2 to your PC and run the DYNAMIXEL Wizard 2.0 app to configure the DYNAMIXEL.
Thank you.

WilliamH07, 2022-02-17:
Hello, thanks
The baudrate and ID was correct, but I didn't change the operating mode with the dynamixel wizard, but when I did it's working ! 
I have a last question and I will close the issue, can I control the xl320 and xl430 at the same time with this program ?  
thank you

ROBOTIS-Will, 2022-02-18:
@WilliamH07 
Yes, you can definitely control different DYNAMIXEL as long as the operating voltages are separated, and you have correct control table address when communicating with DYNAMIXEL.
For example, XL-320 runs off of 7.4V, and uses [2 byte Goal Position at address 30](https://emanual.robotis.com/docs/en/dxl/x/xl320/#control-table-of-ram-area).
On the other hand, XL430 runs off of 11.1V, and uses [4 byte Goal Position at address 116](https://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#control-table-of-ram-area).

These variations need to be covered in your program when you are running different types of DYNAMIXEL.
Thank you.

closed 2025-02-13 by yun-goon