ROBOTIS-GIT/DynamixelSDK #551: Input to output motion differences in dynamixel motors

Raj-Kiran127, 2022-02-19:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  3.7.31

2. Which programming language/tool do you use?
  MATLAB R2021b

3. Which operating system do you use?
  Windows 10

3. Which USB serial converter do you use?
  U2D2

4. Which DYNAMIXEL do you use?
  MX-28(2.0)AT

5. Have you searched the issue from the closed issue threads?
  Yes

6. Please describe the issue in detail
  I am using 3 MX-28AT (2.0) motors connected in a daisy chain. For a desired output (which is attached in the image) I have written a code(attached as well) in which there is some lag between the desired output and the output generated by the motors.
Is it MATLAB issue ? Or communication speed issue ? 
If not, can you please tell me what might be the problem am facing here?
(Code of sync-read write is used as base to write my code. ) 

7. How can we reproduce the issue?
  
clc;
clear all;
close all ;

lib_name = '';

if strcmp(computer, 'PCWIN')
  lib_name = 'dxl_x86_c';
elseif strcmp(computer, 'PCWIN64')
  lib_name = 'dxl_x64_c';
elseif strcmp(computer, 'GLNX86')
  lib_name = 'libdxl_x86_c';
elseif strcmp(computer, 'GLNXA64')
  lib_name = 'libdxl_x64_c';
elseif strcmp(computer, 'MACI64')
  lib_name = 'libdxl_mac_c';
end

% Load Libraries
if ~libisloaded(lib_name)
    [notfound, warnings] = loadlibrary(lib_name, 'dynamixel_sdk.h', 'addheader', 'port_handler.h', 'addheader', 'packet_handler.h', 'addheader', 'group_sync_write.h', 'addheader', 'group_sync_read.h');
end

% Control table address
ADDR_PRO_TORQUE_ENABLE          = 64;                 % Control table address is different in Dynamixel model
ADDR_OPERATING_MODE             = 11;
ADDR_PRO_PROFILE_VELOCITY       = 112;
ADDR_PRO_GOAL_POSITION          = 116;
ADDR_PRO_PRESENT_POSITION       = 132;

% Data Byte Length
LEN_PRO_GOAL_POSITION       = 4;
LEN_PRO_PRESENT_POSITION    = 4;

% Protocol version
PROTOCOL_VERSION            = 2.0;          % See which protocol version is used in the Dynamixel

% Default setting
DXL1_ID                     = 1;            % Dynamixel#1 ID: 1
DXL2_ID                     = 2;            % Dynamixel#2 ID: 2
[... 296 lines cut ...]
dxl_error = getLastRxPacketError(port_num, PROTOCOL_VERSION);
if dxl_comm_result ~= COMM_SUCCESS
    fprintf('%s\n', getTxRxResult(PROTOCOL_VERSION, dxl_comm_result));
elseif dxl_error ~= 0
    fprintf('%s\n', getRxPacketError(PROTOCOL_VERSION, dxl_error));
end

% Close port
closePort(port_num);

% Unload Library
unloadlibrary(lib_name);

figure(1)
movegui('east')
plot(time,M1,'-',time,d1,'x',time,M2,'.',time,d2,'s',time,M3,'-.',time,d3,'o')
ylabel('Input Positions')
xlabel('Time in Second')
legend('Desired-Output-M1','Motor-Output-M1','Desired-Output-M2','Motor-Output-M2','Desired-Output-M3','Motor-Output-M3')

where, 
M1,M2,M3 are desired positions to motors
d1,d2,d3 are output positions from motors

[image]

ROBOTIS-Will, 2022-02-23:
Hi @Raj-Kiran127 

Usually the USB device introduces the latency which is set to 16ms by default in the OS.
In order to minimize the latency, please adjust the USB latency as below.
https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#usb-latency-setting

Thanks.

yun-goon, 2025-02-13:
It looks like the issue has been resolved, so I will close it now.

closed 2025-02-13 by yun-goon