ROBOTIS-GIT/DynamixelSDK #552: Hi all,

skumar648, 2022-03-01:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  - ex) `3.4.1`

2. Which programming language/tool do you use?
  - ex) `C`, `C++`, `C#`, `Python`, `Java`, `MATLAB`, `LABVIEW`, `ROS`, `Arduino`, etc

3. Which operating system do you use?
  - ex) `Ubuntu 16.04`, `Windows 10`, `OS X 10.14`

3. Which USB serial converter do you use?
  - ex) `USB2Dynamixel`, `U2D2`, `OpenCM904`, `OpenCR`, etc

4. Which DYNAMIXEL do you use?
  - ex) `MX-28(2.0)`, `AX-12W`, `XM430-W250`, etc.

5. Have you searched the issue from the closed issue threads?

6. Please describe the issue in detail

7. How can we reproduce the issue?

skumar648, 2022-03-01:
Kindly help as soon as possible

ROBOTIS-Will, 2022-03-07:
@skumar648 
Sorry about the late reply.
Please refer to the original community thread below.
Since you are using MX(2.0) firmware, you may try with the X series examples.
https://community.robotis.us/t/problem-associated-with-labview-code-for-running-the-motor-mx-28ar/702/8?u=willson

yun-goon, 2025-02-13:
Closing this thread as there isn't recent activity. Please feel free to reopen when necessary.

closed 2025-02-13 by yun-goon