ROBOTIS-GIT/DynamixelSDK #554: HI all, can I use DYNAMIXEL SDK to realtime control the ROBOTIS GP using python, with wireless communication using BT-410 and BT-410 dongle?

KivaOP, 2022-03-24:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  - ex) `3.4.1`

2. Which programming language/tool do you use?
  - ex) Python

3. Which operating system do you use?
  - ex) Windows 10

3. Which USB serial converter do you use?
  - ex) I want to use it wireless, which is bluetooth BT-410

4. Which DYNAMIXEL do you use?
  - ex) AX-12A, AX-18A

5. Have you searched the issue from the closed issue threads?

6. Please describe the issue in detail
  I want to using python to realtime control from my computer to my robotis GP through the wireless communication is BT-410 and BT-410 dongle. But I can't find the way to control each DYNAMIXEL without USB serial (except of R+ motion and R+ task). Please help me to solve this problem. Thank you everyone!!!

8. How can we reproduce the issue?

yun-goon, 2025-07-11:
Sorry for the late response. This issue was created a long time ago.

Please try again with the latest version, and if you still encounter any problems, feel free to open a new issue.

Thank you for your understanding!

closed 2025-07-11 by yun-goon

KivaOP, 2025-07-11:
To whom may concern, definitely you can program the Dynamixel through bluetooth module.
The diagram:
Your computer <-> BT-410 <-> BT-410 (connect with MCU) <-> MCU (sth like CM-530) <-> Dynamixel.

Remember to re-program MCU to get the information from BT-410

If you have any problem, you can contact me via email huynhthinh24@gmail.com. I will try to support.