ROBOTIS-GIT/DynamixelSDK #559: programing XM430-W350-R python Current-based Position Control Mode

barak720, 2022-04-27:
1. Which DYNAMIXEL SDK version do you use?
 DynamixelSDK-3.7.31

2. Which programming language/tool do you use?
Python

3. Which operating system do you use?
 Windows

3. Which USB serial converter do you use?
U2D2

4. Which DYNAMIXEL do you use?
  XM-430-W350-R

Dear all, I'm new to python and Dynamixel.
I bought XM430-W350 to use it in the Current-based Position Control Mode.
I looked for the syntacs to use this mode through the SDK test libary, github and generally online,
Unfortunately I havn't found a simple Dynamixel SDK code that uses this mode.
Is there any exaple code for that, lists of syntacs?
thanks in advance.

ROBOTIS-Will, 2022-04-29:
Hi @barak720,

First of all, the Operating Mode of DYNAMIXEL can be easily configured with [DYNAMIXEL Wizard 2.0](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/) which is a recommended GUI tool for managing and testing DYNAMIXEL.
Using this tool will help you to understand better about the control table of DYNAMIXEL.

The Operating Mode of XM430-W350 can be updated by writing a value to the [control table address 11](https://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#operating-mode11), and in your case, you can write `5` to configure as Current-based Position Control Mode.

The Current-based Position Control Mode is similar to the Position Control Mode, but supports [additional current controller](https://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#goal-current102).

If you are trying to write a python code to configure a DYNAMIXEL to run with desired control mode, you may add below code in the read_write.py example before [enabling DYNAMIXEL Torque in line 143](https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/python/tests/protocol2_0/read_write.py#L143).

```python
# Set Current-based Position Control mode
# Modifying EEPROM area value should be done before enabling DYNAMIXEL torque.
# Address of Operating Mode : 11
# Current-based Position Control mode value : 5
dxl_comm_result, dxl_error = packetHandler.write1ByteTxRx(portHandler, DXL_ID, 11, 5)
if dxl_comm_result != COMM_SUCCESS:
    print("%s" % packetHandler.getTxRxResult(dxl_comm_result))
elif dxl_error != 0:
    print("%s" % packetHandler.getRxPacketError(dxl_error))
else:
    print("DYNAMIXEL has been successfully configured as Current-based Position Control Mode.")
```

barak720, 2022-05-20:
Hi @ROBOTIS-Will 
thank you for your reply and sorry for my late one.
The mode works

Thanks alot!

yun-goon, 2025-02-13:
It looks like the issue has been resolved, so I will close it now.

closed 2025-02-13 by yun-goon