ROBOTIS-GIT/DynamixelSDK #560: sync_read_write.c example failure: there is no status packet!

jackbackrack, 2022-05-04:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  - ex) `3.4.1`

2. Which programming language/tool do you use?
  - ex) `C`, `C++`, `C#`, `Python`, `Java`, `MATLAB`, `LABVIEW`, `ROS`, `Arduino`, etc
C

3. Which operating system do you use?
  - ex) `Ubuntu 16.04`, `Windows 10`, `OS X 10.14`
OS X 12.01

3. Which USB serial converter do you use?
  - ex) `USB2Dynamixel`, `U2D2`, `OpenCM904`, `OpenCR`, etc
U2D2

4. Which DYNAMIXEL do you use?
  - ex) `MX-28(2.0)`, `AX-12W`, `XM430-W250`, etc.
XM430-W350

5. Have you searched the issue from the closed issue threads?
yes

6. Please describe the issue in detail
when I run the sync_read_write.c example I get:

  [TxRxResult] There is no status packet!
  [ID:020] groupSyncRead getdata failed

it appears it fails to successfully perform groupSyncReadTxRxPacket and actually catches that failure in the subsequent checks.

I've changed the control table addresses to:

  // Control table address
  #define ADDR_PRO_TORQUE_ENABLE          64
  #define ADDR_PRO_GOAL_POSITION          116
  #define ADDR_PRO_PRESENT_POSITION       132

and am using 57600 baud and serial port "/dev/tty.usbserial-FT6RW6SF".
I'm using two dynamixel with id's 20 and 21.
These settings work for the read_write.c example.
Here's a trace of my execution:

  Succeeded to open the port!
  Succeeded to change the baudrate!
  Dynamixel#20 has been successfully connected 
  Dynamixel#21 has been successfully connected 
  Press any key to continue! (or press ESC to quit!)
  s
  [TxRxResult] There is no status packet!
  [ID:020] groupSyncRead getdata failed

7. How can we reproduce the issue?

here are the changed section of the code but otherwise it's just using the standard sync_read_write.c example:

// Control table address
#define ADDR_PRO_TORQUE_ENABLE          64                 // Control table address is different in Dynamixel model
#define ADDR_PRO_GOAL_POSITION          116
#define ADDR_PRO_PRESENT_POSITION       132

// Data Byte Length
#define LEN_PRO_GOAL_POSITION           4
#define LEN_PRO_PRESENT_POSITION        4

// Protocol version
#define PROTOCOL_VERSION                2.0                 // See which protocol version is used in the Dynamixel

// Default setting
#define DXL1_ID                         20                   // Dynamixel#1 ID: 1
#define DXL2_ID                         21                   // Dynamixel#2 ID: 2
#define BAUDRATE                        57600
#define DEVICENAME                      "/dev/tty.usbserial-FT6RW6SF"
                                                            // ex) Windows: "COM1"   Linux: "/dev/ttyUSB0" Mac: "/dev/tty.usbserial-*"

#define TORQUE_ENABLE                   1                   // Value for enabling the torque
#define TORQUE_DISABLE                  0                   // Value for disabling the torque
#define DXL_MINIMUM_POSITION_VALUE      0                   // Dynamixel will rotate between this value
#define DXL_MAXIMUM_POSITION_VALUE      4095                // and this value (note that the Dynamixel would not move when the position value is out of movable range. Check e-manual about the range of the Dynamixel you use.)
#define DXL_MOVING_STATUS_THRESHOLD     20                  // Dynamixel moving status threshold

#define ESC_ASCII_VALUE                 0x1b

ROBOTIS-David, 2022-05-06:
@jackbackrack 

Hi, 

To reproduce your issue, I ran the basic code of [sync_read_write.c](https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/c/example/protocol2.0/sync_read_write/sync_read_write.c) by only modifying Control Table config as below, which is same aside from `DXL#_ID ` and `DEVICENAME` (Ubuntu 16.04)

```C
// Control table address
#define ADDR_PRO_TORQUE_ENABLE          64                 // Control table address is different in Dynamixel model
#define ADDR_PRO_GOAL_POSITION          116
#define ADDR_PRO_PRESENT_POSITION       132

// Data Byte Length
#define LEN_PRO_GOAL_POSITION           4
#define LEN_PRO_PRESENT_POSITION        4

// Protocol version
#define PROTOCOL_VERSION                2.0                 // See which protocol version is used in the Dynamixel

// Default setting
#define DXL1_ID                         1                   // Dynamixel#1 ID: 1
#define DXL2_ID                         2                   // Dynamixel#2 ID: 2
#define BAUDRATE                        57600
#define DEVICENAME                      "/dev/ttyUSB0"      // Check which port is being used on your controller
                                                            // ex) Windows: "COM1"   Linux: "/dev/ttyUSB0" Mac: "/dev/tty.usbserial-*"

#define TORQUE_ENABLE                   1                   // Value for enabling the torque
#define TORQUE_DISABLE                  0                   // Value for disabling the torque
#define DXL_MINIMUM_POSITION_VALUE      0             // Dynamixel will rotate between this value
#define DXL_MAXIMUM_POSITION_VALUE      4095              // and this value (note that the Dynamixel would not move when the position value is out of movable range. Check e-manual about the range of the Dynamixel you use.)
#define DXL_MOVING_STATUS_THRESHOLD     20                  // Dynamixel moving status threshold

#define ESC_ASCII_VALUE                 0x1b

```

My DXL setup (ID1, ID2, Position Control Mode, 57600 bps)  

[image]

And it works perfrectly without any issue. 

[image]

Therefore,

1. Be sure to re-build your make file after changing the code.
2. Be sure to make the corresponding baudrate to comm at the same protocol.

jackbackrack, 2022-05-06:
I confirmed that my code works on linux with my dynamixel xm430's but doesn't work on my mac laptop.  I 1. rebuilt the code from scratch with make and 2. checked that the baudrate is correct at 57600.  In summary, the same dynamixels and code work with sync_read_write on linux but not on the mac.  The only change I made between platforms was the serial port pathname.  Note that this all works fine on the mac with the regular read_write.c example just not the sync_read_write.c example.

jackbackrack, 2022-05-06:
OK, I figured it out.  The brand new USB-C to USB-A adaptor didn't work.  I used a mac one and suddenly it all works.  That was subtle.  Sorry for your trouble!

jackbackrack, 2022-05-08:
Unfortunately, now even the Apple USB-C to USB-A adaptor doesn't work.  I can't get the sync_read_write.c example to work at all.  I have an M1 MacBook.  I'm not sure if that has anything to do with it but it fails in the same way by not being able to read the status packet.

ROBOTIS-David, 2022-05-09:
Well, unfortunately, I have had no chance to test M1 macbook with U2D2. Hope that I can find one and attempt.. 

1. How about the other SDK examples? Does them all work without any issue? Seems that `sync_read_write.c` is the only  issue in your case, looking at the context above.

2. Can you try [DYNAMIXEL Wizard 2.0](https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/#packet) and do Packet? 

You can perform DYNAMIXEL instruction with the port connected.  And see any issue occur as. 

[image]

jackbackrack, 2022-05-09:
ok, i tried the wizard and had no trouble with ping read and sync read.  I've only tried read-write.c and sync-read-write.c examples.  read-write.c worked fine but sync-read-write.c fails repeatedly.
[image]

ROBOTIS-David, 2022-05-12:
That is so weired. 

I have my own M1 mackbook at home. Later I will go test and see if any issue I can find.

yun-goon, 2025-02-13:
Closing this thread as there isn't recent activity. Please feel free to reopen when necessary.

closed 2025-02-13 by yun-goon