ROBOTIS-GIT/DynamixelSDK #569: unable to build package

sardanian, 2022-06-25:
I am trying to colcon build the foxy2 branch to use in the open_manipulator.

To recreate:
cd ~/colcon_ws/src
git clone -b foxy-devel https://github.com/ROBOTIS-GIT/DynamixelSDK.git
cd ~/colcon_ws
colcon build

This is my error:
Starting >>> dynamixel_sdk
Starting >>> dynamixel_sdk_custom_interfaces
Finished <<< dynamixel_sdk [1.23s]                                     
--- stderr: dynamixel_sdk_custom_interfaces                             
make[1]: *** No rule to make target 'dynamixel_sdk_custom_interfaces__py/CMakeFiles/dynamixel_sdk_custom_interfaces__py.dir/all', needed by 'CMakeFiles/dynamixel_sdk_custom_interfaces__python.dir/all'.  Stop.
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:146: all] Error 2
---
Failed   <<< dynamixel_sdk_custom_interfaces [1.56s, exited with code 2]

Summary: 1 package finished [1.76s]
  1 package failed: dynamixel_sdk_custom_interfaces
  1 package had stderr output: dynamixel_sdk_custom_interfaces
  1 package not processed

this used to work for me a few days ago but I seem to be having issues now with it even though I have a fresh workspace.

thehummingbird, 2022-10-02:
Hi,
Were you able to fix this? I have similar issues

yun-goon, 2025-02-13:
Apologies for the delayed response.  We kindly ask that you try again with the latest version. If the same issue arises, please feel free to resubmit it, and we will address it promptly.

closed 2025-02-13 by yun-goon