ROBOTIS-GIT/DynamixelSDK #571: Position Controller Feedforward Calculation

mrunaljsarvaiya, 2022-06-29:
For a XM540 in position controller mode, could I get an explanation of how the first feedforward term is calculated (Referring to the block diagram at 2.4.25 at https://emanual.robotis.com/docs/en/dxl/x/xl430-w250/). Is the feedforward based on the present motor position and goal position or does it use the previous goal position and the new incoming goal position to calculate a feedforward velocity?

ROBOTIS-David, 2022-07-19:
As the image describes, the Feedforward refers to the generated profiles ([Velocity Trajectory(136)](https://emanual.robotis.com/docs/kr/dxl/x/xl430-w250/#velocity-trajectory136), [Position Trajectory(140)](https://emanual.robotis.com/docs/kr/dxl/x/xl430-w250/#position-trajectory140))

[image]

Make sure that internal controller gains should be determined by using experiment approach. 

See - [DYNAMIXEL Tips | DYNAMIXEL Position Controller PID Gain Tuning (feat. Arduino example)](https://www.youtube.com/watch?v=msWlMyx8Nrw)

yun-goon, 2025-02-13:
Closing this thread as there isn't recent activity. Please feel free to reopen when necessary.

closed 2025-02-13 by yun-goon