ROBOTIS-GIT/DynamixelSDK #572: Hi, I need some assistance, I have the XM540-W150R and was wondering how to set the goal positions above 4095 and below 0 I need them to continuously turn until it hits a goal position?

B3n77, 2022-07-07:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use?
  - ex) `3.4.1`

2. Which programming language/tool do you use?
  - ex) `C`, `C++`, `C#`, `Python`, `Java`, `MATLAB`, `LABVIEW`, `ROS`, `Arduino`, etc

3. Which operating system do you use?
  - ex) `Ubuntu 16.04`, `Windows 10`, `OS X 10.14`

3. Which USB serial converter do you use?
  - ex) `USB2Dynamixel`, `U2D2`, `OpenCM904`, `OpenCR`, etc

4. Which DYNAMIXEL do you use?
  - ex) `MX-28(2.0)`, `AX-12W`, `XM430-W250`, etc.

5. Have you searched the issue from the closed issue threads?

6. Please describe the issue in detail

7. How can we reproduce the issue?

ROBOTIS-David, 2022-07-19:
Hi @B3n77 
Can not find any information available. Make sure to fill the templete properly.

ASErkek, 2022-12-15:
Maybe you can use extended position mode

yun-goon, 2025-02-13:
Closing this thread as there isn't recent activity. Please feel free to reopen when necessary.

closed 2025-02-13 by yun-goon