ROBOTIS-GIT/DynamixelSDK #574: destruction data for XL320

yuya-0411, 2022-07-11:
@ROBOTIS-Will 
Hello, I'm Yuya from Japan.

Please fill this template for more accurate and prompt support.

4. Which DYNAMIXEL do you use?
  - XL-320
 
5. Have you searched the issue from the closed issue threads?
- Yes. But I didn't find the issue.

7. Please describe the issue in detail
- I would like to know the data for XL-320. In my study, I would like to know how much impact force can break the XL-320 and the bracket connected to it. If you don't have the data, I am going to implement some examinations in order to get the data.

8. How can we reproduce the issue?
If you have the data related to destruction data for XL-320 and the bracket, could you teach me it?

Thank you for your attention.

ROBOTIS-Will, 2022-07-14:
Hi @yuya-0411 

Thank you for your inquiry.
Although I'll reach out to the relevant development team for this information, they are very likely not have actual test result as XL320 is not designed for a harsh environment.
Thank you.

yuya-0411, 2022-07-14:
@ROBOTIS-Will 
Thank you for your rapid reply.
I'm interested in that development.

I can understand that XL320 is not designed for a harsh environment.

I would like to know it for destruction simulation. 
Actually, I simulate my robot consisting of XL320 and would like to know tips for the definition of destruction.
If you have or find the tips, could you teach me them? Just a tip is ok. 

Thank you for your time.

ROBOTIS-Will, 2022-07-18:
Hi @yuya-0411 

Unfortunately, due to the nature of the product, we do not have any simulation or test data for the product.
The XL320 has prominent brakets for assembly to robot frames which makes it less robust.
In case where you need more solid robot body, i think XL330 or XC330 would be a better choice.

Thank you very much.

yun-goon, 2025-02-13:
It looks like the issue has been resolved, so I will close it now.

closed 2025-02-13 by yun-goon