ROBOTIS-GIT/DynamixelSDK #575: Unistall dynamixel sdk on python Linux to replace it with an other version

alinena, 2022-07-13:
ISSUE TEMPLATE ver. 1.2.0

1. Which DYNAMIXEL SDK version do you use?
  - 3.7.31

2. Which programming language/tool do you use?
  - , `Python`,  `ROS`,

3. Which operating system do you use?
  -`Ubuntu 18.04.6`

3. Which USB serial converter do you use?
  -  `USB2Dynamixel`

4. Which DYNAMIXEL do you use?
  -`MX-28AT`

5. Have you searched the issue from the closed issue threads?yes

6. Please describe the issue in detail
I would like to unistall the dynamixel-sdk 3.7.31 to install an other version that may be would be  better to resolve some issues.
Does anyone know a way to do it on python , linux ?

ROBOTIS-David, 2022-07-19:
Hi, @alinena 

You can simply go back to commit history by doing hard reset. The following shows the Git histroy, the function provdided by Visual Studio. Alternatively, you can also find git histroy usig "git log" + any option such as --history. 

[image]

yun-goon, 2025-02-13:
It looks like the issue has been resolved, so I will close it now.

closed 2025-02-13 by yun-goon