ROBOTIS-GIT/DynamixelSDK #576: FastGroupSyncRead functionality no longer working

CImogu, 2022-07-29:
ISSUE TEMPLATE ver. 1.2.0

Please fill this template for more accurate and prompt support.

1. Which DYNAMIXEL SDK version do you use? 3.7.31

2. Which programming language/tool do you use? Python

3. Which operating system do you use? OS Monterey 12.5

3. Which USB serial converter do you use? U2D2

4. Which DYNAMIXEL do you use? XL-320

5. Have you searched the issue from the closed issue threads? Yes

6. Please describe the issue in detail
- After reading through #527, I tested the FastGroupSyncRead feature implemented in the develop branch. At first, it worked amazingly and was way faster than the regular GroupSyncRead.
- But a few hours later when testing the code again, the isAvailable method of the new GroupSyncRead class stopped working. My addresses are correct, but somehow the fastSyncReadRx method of the packet handler does not communicate successfully, but the fastSyncRead of GroupSyncRead works great.

7. How can we reproduce the issue?
- The issue arose when testing [the fast_sync_read.py file at https://github.com/ROBOTIS-GIT/DynamixelSDK/tree/develop](https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/84723f86b0643284af1054d3afb515b23da1e328/python/tests/protocol2_0/fast_sync_read.py)

CImogu, 2022-07-29:
I also made sure to copy the dynamixel_sdk to the folder mentioned above to make sure it was not using the python library instead which does not yet support fast sync read.

yun-goon, 2025-03-14:
Hello @CImogu 

The FastBulkRead or FastSyncRead Python functionality has been updated to the main branch! 🚀

Feel free to check it out, and let me know if you have any questions!

closed 2025-04-18 by yun-goon